SpenceKonde/ATTinyCore #322: Attiny 85 no funciona Tone() con core 114

paezdaniel, 2019-06-06:
Cuando finaliza el tono, la salida en algunas ocasiones queda en alto,  y no se puede vover a escrbir con digtalWrite

SpenceKonde, 2019-06-07:
By 114, do you mean ATTinyCore 1.1.4? That version is really old - check with 1.2.4. I made a number of changes to Tone() on tiny85 for recent versions, this may already be fixed. If it still happens with 1.2.4 let me know - the code for Tone() is really messy, and I wouldn't be surprised if there was an issue here.

paezdaniel, 2019-06-07:
Muchas gracias por la rapida respuesta..
Perdon quise escribir 1.2.4
AttATTinyCore 1.2.2  funciona perfecto
AttATTinyCore 1.2.4  tiene errores... Una vez llamada la funcion tone( ) no se puede volver a usar digitalWrite. .. no funciona...
Ademas.  cuando termina el tono la salida queda en HIGH,   cuando deberia quedar en LOW

SpenceKonde, 2019-06-07:
Which pin are you using?

SpenceKonde, 2019-06-07:
Something tells me it's pin 3.

paezdaniel, 2019-06-07:
pin 6 de attiny 85  o sea.. D1

SpenceKonde, 2019-06-07:
Oh. hmm

SpenceKonde, 2019-06-07:
Oh wait, no - that makes sense. I remembered wrong, pin 1 and 4 are the ones that use hardware output compare for Tone(), not 3 and 4. And I added pin 1 support for hardware output compare in 1.2.4. 

Does pin 4 work, or does it do the same thing?

paezdaniel, 2019-06-07:
No lo he probado en pin 4, para mañana lo verifico y te cuento

SpenceKonde, 2019-06-07:
Aaaha!! 

I forgot to make TurnOffPWM() account for the weird situation on pin 1 - which uses timer0 for normal PWM, but timer1 for tone()! But since digitalPinToTimer() only checks digital_pin_to_timer_PGM[], it turns off timer0's channel, and doesn't touch timer1, so the timer is left connected to the pin with the clock stopped.

SpenceKonde, 2019-06-07:
May also be broken on tiny43.

paezdaniel, 2019-06-07:
> Aaaha !!
> 
> Olvidé (o más bien, no me di cuenta de que necesitaba) hacer que TurnOffPWM () tuviera en cuenta la situación extraña en el pin 1, que usa timer0 para PWM normal, pero timer1 para tone (). Pero como digitalPinToTimer () solo comprueba digital_pin_to_timer_PGM [], desactiva el canal del timer0, y no toca timer1, por lo que el temporizador se queda conectado al pin con el reloj parado.

Muy dificil para mi jaja;
Entonces no necesita que haga ninguna prueba?

paezdaniel, 2019-06-07:
Aclaro que el tono si se reproduce,  el problema es al finalizar el tono que queda en HIGH, cuando deberia quedar en LOW  y que ademas queda en ese estado HIGH y digitalWrite ( 1, low) no funciona

closed 2019-06-10 by SpenceKonde

SpenceKonde, 2019-06-10:
Should be good now in the github version. Fix will be in 1.2.5 when it is released to board manager - but I don't know when that will be, since I just released 1.2.4. 

To make it work with 1.2.4 an 1.2.3 (the bug was introduced in 1.2.3), after stopping the tone, do:
TCCR1&=207;

paezdaniel, 2019-06-11:
Muchas gracias  esperare sin apuro la version 1.2.5  por ahora estoy usando la 1.2.2 y funciona perfecto

commit b940a523 (2019-06-10): Fix #322

Need to turn off TIMER1A PWM, because pin 1 analogWrite() pwm is controlled by TIMER0A, and so digitalWrite() won't disconnect the timer from the pin.
commit 3adc457d (2019-06-10): Fix #322 correctly

Previous fix didn't compile, and would have been inferior even if it did.