SpenceKonde/ATTinyCore #323: ATtiny841 SD Card Compile Error - CardInfo Example

altech6983, 2019-06-08:
Trying to get an sd card working with my 841 but the CardInfo Example won't compile.

`Arduino: 1.8.9 (Windows 10), Board: "ATtiny441/841 (No bootloader), ATtiny841, Disabled, 8 MHz (internal), EEPROM retained, B.O.D. Enabled (1.8v), B.O.D. Disabled, B.O.D. Disabled, Counterclockwise (like old ATTinyCore and Rev. C boards), Master Only"`

`C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:457:19: error: 'PORTC' was not declared in this scope   {&DDRC, &PINC, &PORTC, 5}   // C5 19`

`Using library SPI at version 1.0 in folder: C:\Users\Coldfission\Documents\Arduino\hardware\ATTinyCore\avr\libraries\SPI 
Using library SD at version 1.2.3 in folder: C:\Program Files (x86)\Arduino\libraries\SD 
exit status 1
Error compiling for board ATtiny441/841 (No bootloader).`

I shortened that a bit because there were a whole bunch about PORTC and PORTD. I assume I am doing something wrong. I traced the problem to I think a lack of `pinmap attiny841` in the Sd2PinMap.

Do I need a special sd library for atttinycore? I assumed it was just the included SPI that needed to be different.

Thank you for any help.

SpenceKonde, 2019-06-08:
The SPI library supplied with the core does indeed provide spi
functionality (but for the 841, this is no different from the normal spi
library- the 841 has a real spi port).

But the sd card library doesnt support the attinys. As I understand it, the
limited ram of the tinies is the real problem here. As sd cards are written
in 512b sectors, which must be buffered in ram, a microcontroller with only
512b of ram cant do that.... a bit of searching didnt turn up a good
solution for the much more popular tiny85, which has the same 512b of ram.

If you need lots of nonvolatile memory on an attiny841, I think you may be
limited to external eeprom chips (24-series for i2c and 25-series for spi)
and spi flash chips... both of which are not nearly as convenient if you
hope to get the data onto a computer (though there is a cheap usb
reader/writer for them for a few bucks on ebay)

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

sleemanj, 2019-06-08:
Hmm.  I wonder if there is any reason you can't use the EEPROM as the necessary buffer space.  Or for that matter some reserved chunk of flash.  Maybe not fast enough?

altech6983, 2019-06-08:
I totally forgot about the 512b requirement. I kind of assumed that the library would allow read only in parts but why would it, it was never made for this small of ram. 

There is a solution to my specific problem (and help for anyone that finds this thread)

[Petit FAT](http://elm-chan.org/fsw/ff/00index_p.html)
[Arduino Wrapper For Petit FAT](https://github.com/greiman/PetitFS)

I should have mentioned that I only need to read a wav file and feed it to TMRpcm library.

At this point though, the board that I designed has some problems and needs a v2 anyways so I might as well just use one of those [HiLetgo Arduino MP3 Player Mini Modules](https://www.amazon.com/gp/product/B01D1D0E7Q/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1) and stop with the hassle and just use the tiny to control it and the rest of the project.

Thank you for the help and quick response.

closed 2019-06-08 by altech6983