SpenceKonde/ATTinyCore #324: possibility to swap MISO/MOSI or i2c SDA pins

chocotov1, 2019-06-19:
I'm using SPI and i2c on a ATtiny85 and the ATTinyCore. In my project i've been able to use a Lora SPI module and i2c LCD module together.

As i'm only sending data to the Lora module, i only need MOSI / DO / pin 6.

The i2c SDA is on DI / pin 5 .

I was wondering if it would be possible to either put MOSI on pin 5, or SDA on pin 6 in an effort to free up one pin for general use.

I've experimented a bit in the file avr/variants/tinyX5/pins_arduino.h, but for instance MOSI would stay on the same pin regardless of the changes i made.

SpenceKonde, 2019-06-19:
No, this cannot be done - the I2C and SPI pins are fixed by the locations of the DI, DO, and USCK pins (USI pins); the #defines in the variant file are used to adjust pinMode on these pins and to maintain compatibility with libraries and code that rely on being able to read/write/set mode on these pins by their names. You cannot move them around - this is fixed by the hardware. 

Both I2C and SPI on the ATtiny parts with a USI use the same hardware peripheral for these tasks (that's why they use the same pins, and another reason why you couldn't use both at the same time.

You would need to use a software I2C implementation (ex. SoftI2CMaster) and modify the library for the I2C device to work with this library.

closed 2019-06-19 by SpenceKonde

chocotov1, 2019-06-19:
Thanks for the feedback. I'll have look at SoftI2CMaster.

SpenceKonde, 2019-06-19:
If it's the I2C screen that uses LiquidCrystal_I2C library (the 1602/2004 character LCDs), https://github.com/SpenceKonde/SoftI2CMaster and https://github.com/SpenceKonde/LiquidCrystal_I2C_Tiny may do what you want with little if any modification, because back before I did the Universal Wire library, I had to make an LCD work with a t1634.

chocotov1, 2019-06-20:
Thanks for the tips! At the moment i'm using the LiquidCrystal_I2C library with the 1602 screen.