SpenceKonde/ATTinyCore #326: Need help for programming Attiny4313

EmerickH, 2019-07-13:
Hello,
I'm trying to program an Attiny4313 but I'm encountering some weird issues...
For testing, I'm using this little code:
```
void setup() {
  Serial.begin(9600);
  pinMode(7, OUTPUT);
}

void loop() {
  digitalWrite(7, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(500);                       // wait for a second
  Serial.println("Hey!");
  digitalWrite(7, LOW);    // turn the LED off by making the voltage LOW
  delay(500);                       // wait for a second
  Serial.println("Hey!");
}
```
1.  The LED on pin 7 is blinking but way to fast (maybe at ~100ms), and if I put 1s delays, it's taking ~2-3s between each change! Maybe a clock issue? I'm using internal clock at 8Mhz, but I'm planning to use an external 10Mhz one in future.
2. When I plug power, the LED starts blinking only one in two, I need to reset/power cycle to make it work! Sometimes, the LED also stops blinking for a few seconds and then restarts to blink...
3. Nothing is sent on the serial (I tested with different methods and double checked I'm using the right pin)
It's maybe just a simple problem but I can't find it! I hope you can help me because I'm a little bit lost!

Thank you,
Emerick

SpenceKonde, 2019-07-13:
1. Timing anomalies generally indicate that the clock is not running at the speed you think it is. You must do "Burn Bootloader" to set the fuses for the selected clock speed after selecting a clock speed option - normal upload via ISP doesn't set the fuses (this is the same way it works on the official Arduino AVR core - that's why I have it work the same way on my core). 

2. Sounds like it's resetting. Did you forget the 0.1uF ceramic capacitor between Vcc and Gnd on the attiny4313 (located as close to the pins as possible)? Without this, the chip may reset or fail to operate at random; unfortunately many guides omit this required component - the behavior without the decoupling cap can depend on the individual chip (process variation), stability of power supply, length of wires between power supply and chip, and the code and what is connected to the pins. Thus, for some simple cases with short wires and the like, it may appear to work fine, while in others, or with other individual chips, it may not. Do not trust any guide that shows connections without including the decoupling cap - that is proof that the author of said guide is not qualified to be writing a guide and may have made other errors. This decoupling cap is required for all digital IC's unless the datasheet specifies otherwise. 

Also check physical connections for loose wires and such. 

3. If the clock speed is wrong, you would see nothing or gibberish on the serial port, because the baud rate would be far off from what it's supposed to be.

EmerickH, 2019-07-14:
Thank you for your help,
I've already burnt the bootloader at 8Mhz to set fuses, but I just tested to burn it to 4Mhz and it seems that timings are better, even if I still don't have any serial output...
I also have the 0.1uF capacitor (that's a multilayer one, it's the only one I have, but I think it's not the problem?), I just connected it differently to be closer and now the chip is starting everytime, but it still stopping sometimes...
The strange thing is serial is not even outputting at wrong baudrate on TX pin, I get a constant 0V output with an oscilloscope...
I tried to do an OSCAL calibration, but I can't even get any PWM signal to test calibration...

EmerickH, 2019-07-14:
Ok for serial, it was a defective chip! After swapping to a new one everything seems to be OK!
Thank you for your help!

EmerickH, 2019-07-14:
Ok now I have another issue!
I can't receive anything on UART...
I even tested with datasheet examples:
```C
void setup(){
  OSCCAL = 86;  
  USART_Init(9600);
 }

void loop(){
  unsigned char temp = USART_Receive();
  USART_Transmit(temp);
}

unsigned char USART_Receive(){
  /* Wait for data to be received */
  while ( !(UCSRA & (1<<RXC)) );
  /* Get and return received data from buffer */
  return UDR;
}

void USART_Transmit( unsigned char data ){/* Wait for empty transmit buffer */while ( !( UCSRA & (1<<UDRE)) );/* Put data into buffer, sends the data */UDR = data;}

void USART_Init( unsigned int baud ){
  /* Set baud rate */
  UBRRH = (unsigned char)(baud>>8);
  UBRRL = (unsigned char)baud;
  /* Enable receiver and transmitter */
  UCSRB = (1<<RXEN)|(1<<TXEN);
  /* Set frame format: 8data, 2stop bit */
  UCSRC = (3<<UCSZ0); //(1<<USBS)|
}
```
And nothing...
It's the same problem as here:
https://forum.arduino.cc/index.php?topic=324354.0
But it seems to have been corrected in ATTinyCore...

SpenceKonde, 2019-07-14:
Interesting - while there is a lot of stuff that I haven't tested, I know 100% that my core is capable of receiving data on the serial port of the '4313, because this sketch works for me (I use it to control a Futaba VFD display):
https://github.com/SpenceKonde/AzzyProjects/blob/master/FutabaToSerial/FutabaToSerial.ino

I don't have a '4313 on me (am away for the weekend) but am getting back tonight and will check that a serial test sketch that echos back characters sent to it works correctly for me. 

The fact that you had a serial issue before, where it worked, and then stopped working and you determined the chip to be bad, and replaced and then it worked - and now it's not working - is making me suspicious that there might be something wrong with your hardware configuration that is damaging chips.

EmerickH, 2019-07-15:
Yes for the TX pin I think I damaged it (I know why) but for the RX, I just connect it to the TX pin of an Arduino Uno.
On the datasheet of 4313 it's written that the resonator is calibrated for 3V, so I'm suspecting it's just because I can't tune it with enough precision. I'll receive very quickly some 16Mhz quartz oscillator, I think its going to be better... Do you use one in your project?

EmerickH, 2019-07-16:
Yes!!! With external 16Mhz quartz serial is working well!!!
Thank you again for your help, I hope I will not have to re-open this issue again...

closed 2019-07-16 by EmerickH