SpenceKonde/ATTinyCore #327: x41 INT0 pin definition mixup?

futuranet, 2019-07-15:
Come across a couple of issues porting sketches to the later library including the clockwise pinout, particularly using attachInterrupt + digitalPinToInterrupt .

I believe I have tracked these down to pin definitions which don't match the pinout image/datasheet

https://github.com/SpenceKonde/ATTinyCore/blob/39e7f6f8a01b098180a63c48c8428e9de6befbf3/avr/cores/tinymodern/core_pins.h#L61
https://github.com/SpenceKonde/ATTinyCore/blob/39e7f6f8a01b098180a63c48c8428e9de6befbf3/avr/cores/tinymodern/core_pins.h#L62

Should be swapped:

    #define PIN_B0  ( 10 )
    #define PIN_B1  ( 9 )

Also 

https://github.com/SpenceKonde/ATTinyCore/blob/39e7f6f8a01b098180a63c48c8428e9de6befbf3/avr/cores/tinymodern/core_pins.h#L81

Should be

    #define CORE_INT0_PIN  PIN_B1

As INT0 is on PB1 => PIN_B1 aka  9 (clockwise numbering) / 1 (ccw)

SpenceKonde, 2019-07-15:
Yup. Thanks, will fix this tonight.

closed 2019-07-17 by SpenceKonde

PaulZC, 2019-08-03:
Hi Spence (@SpenceKonde)
Sincere thanks for sharing this work - I am finding it very useful.
@futuranet identified that the 841 CORE_INT0_PIN should be PIN_B1 not PIN_B2 (according to the datasheet at least...)
Does [this line](https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/cores/tinymodern/core_pins.h#L81) need to be corrected too?

[image]

Many thanks,
Paul