SpenceKonde/ATTinyCore #328: Unable to program T841 (Optiboot) using FTDI cable

cjkeithley, 2019-07-16:
I believe that I've successfully programmed the Optiboot bootloader into the T841 board (using a USBtinyISP) given that the T841's LED flashes after burning the bootload, but when I try to upload my sketch using the IDE (1.8.9) and an FTDI TTL cable I get 10 stk500_getsync() errors:

> avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x94
> avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x88
> avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x94
> avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x88
> ...
> avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x88
> An error occurred while uploading the sketch

The board type is set to USBtinyISP (ATTinyCore) when trying to upload the sketch using the FTDI cable.

I am able to upload my sketch into the T841 using the USBtinyISP (board type set to ATTiny841 no bootloader, programmer set to USBtinyISP), and the sketch runs successfully; I'm able to communicate with the T841 via the IDE's serial monitor (my sketch uses serial.read & serial.write.

Anyone have any thoughts or recommendations?  I assume I'm doing something wrong, but I can't figure out what it might be.

Thanks,

cjk

SpenceKonde, 2019-07-16:
Do you have: 
10k pullup resistor from Reset of t841 to Vcc
0.1uF cap between Reset of t841 and DTR of FTDI cable? 

When you start uploading, do you see the LED blink? 

Are you running the board at 4v or more off the internal oscillator? Per documentation:
> The ATtiny x41 series, 1634R, and 828R have an internal oscillator factory calibrated to +/- 2% - but only at operating voltage below 4v. Above 4v, the oscillator gets significantly faster, and is no longer good enough for UART communications. The 1634 and 828 (non-R) are not as tightly calibrated (so they may need tuning even at 3.3v) and are a few cents less expensive, but suffer from the same problem at higher voltages. Due to these complexities, it is recommended that those planning to use serial (except on a x41, 1634R or 828R at 2.5~3.3v) use an external crystal.

cjkeithley, 2019-07-17:
> Do you have:
> 10k pullup resistor from Reset of t841 to Vcc
> 0.1uF cap between Reset of t841 and DTR of FTDI cable?

Yes... it's one of your T841 boards (Rev C). I purchased it through Tindie.

> When you start uploading, do you see the LED blink?

Yes, but the Arduino ID reports 10 stk500_getsync errors and my sketch doesn't run.

I suspect that there's a step I'm getting wrong (perhaps in the board settings) when I program the bootloader using the Arduino IDE and the USBtinyISP, such that when the T841 comes out of reset and starts running the bootloader, the bootloader isn't able to successfully capture serial data as the IDE attempts to transfer the sketch to the T841 via the FTDI cable.

I believe the FTDI cable is working properly, given that my sketch uses hardware serial read & write, and I'm able to interact with my sketch at 115.2Kbaud using the IDE's serial monitor.

Thanks,

-cjk-

SpenceKonde, 2019-07-18:
You said that you're able to interact with the sketch - are you aware that, after uploading a sketch via ISP, the bootloader is erased? You need to re-bootload it after uploading a sketch via ISP.

What configuration (clock source, frequency) and operating voltage are you using? Did you get the board from me in assembled form or as a bare board that you mounted your own part to? 

Make sure that you had the correct clock options selected when you did burn bootloader, and that you still have them set when you upload - any mismatch will result in STK500 sync errors. 

The fact that serial works from a sketch uploaded via ISP, but not for the bootloader, and that autoreset appears to be working, is strange, and strongly suggests the wrong clock speed was selected when bootloader was burned, or something like that.

cjkeithley, 2019-07-20:
> are you aware that, after uploading a sketch via ISP, the bootloader is erased? 
> You need to re-bootload it after uploading a sketch via ISP.

I did not know!  I've been using your board in my prototype, and I hadn't yet decided whether my PCB would include an ISP connector... it will now!   

> What configuration (clock source, frequency) and operating voltage are you using? 
> Did you get the board from me in assembled form or as a bare board that you 
> mounted your own part to?

I got the board assembled.  It's 5V, and IIRC, 16 Mhz..

> The fact that serial works from a sketch uploaded via ISP, but not for the bootloader, 
> and that autoreset appears to be working, is strange, and strongly suggests the 
> wrong clock speed was selected when bootloader was burned, or something like that.

That's what I suspect as well.. Can you take a look at the following steps and let me know if I'm doing anything else wrong?

Connect the T841 board to the USBTinyUSB.
Plug the USBTinyUSB into the computer
In the Arduino IDE...
... set the board to ATTiny441/841 (no bootloader)
... set the chip to ATTiny841
... the clock to 16 MHz external
... set the programmer to "USBTinyISP"
... set LTO and B.O.D. to disabled
... set pin mapping to??? I have a "rev C" T841.  seems like I should I set this to "counterclockwise"

use Tools->program bootloader to program the bootloader into the T841

To program the T841 using the FTDI cable..

disconnect the T841 board from the USBTinyISP
connect the T841 to the FTDI TTL cable
connect the FTDI TTL cable to the computer
in the Arduino IDE leave the settings unchanged, except for:
... set the board to ATTiny441/841 (Optiboot)
... set the programmer to "USBTinyISP (ATTinyCore)"
... set the port to the FTDI TTL 

click on the upload button in the IDE or use Sketch->Upload to upload the sketch using the FTDI cable.

I'm not sure how often I'll use the bootloader/FTDI cable approach, given that after uploading a sketch I have to reprogram the bootloader.  I suspect I'll use the USBTinyISP the vast majority of the time.

Thanks & best regards,

-cjk-

SpenceKonde, 2019-07-20:
When you want to put the optiboot bootloader on the board, you need to select ATTiny441/841 (optiboot) when you do "burn bootloader". If you select (no bootloader), it will clear the flash (no bootloader) and set the fuses for the BOD and clock source. 

You misunderstood about when the bootloader is erased - If you upload a sketch via ISP (that is, by connecting your USBTinyISP and selecting ATtiny841/441 (no bootloader), or Sketch -> upload using programmer when ATtiny841/441 (optiboot) is selected), the bootloader is erased. It is not erased when uploading via a serial adapter.

cjkeithley, 2019-07-20:
> You misunderstood about when the bootloader is erased -

Thanks for clarifying this.  It seemed weird that uploading via FTDI would erase the bootloader.

But, I'm now having another problem.  I tried to program the bootloader a few moments ago and got an invalid device signature error:

> avrdude: Yikes!  Invalid device signature.
>          Double check connections and try again, or use -F to override
>          this check.
> 
> An error occurred while uploading the sketch

I get the same error when trying to upload a sketch.

I did update some of the boards using the board manager earlier today, so that's probably the [indirect] cause.

But, sadly, I'm not able to get back to a working configuration.  I tried removing & reinstalling the ATTiny board package.  No luck, same error message.  Yes, I've double and triple checked the connections between the USBtinyISP and the T841.  

Any thoughts?  If not, I have a couple other T841s that I'll try (after adding some headers).

Thanks,

-cjk-

SpenceKonde, 2019-07-20:
Enable verbose upload in preferences, and attempt to burn bootloader again - it will tell you what signature it actually sees. 

0x000000 indicates missing clock or hardware problem. 

Other signatures - depends what it is. 

Also... make sure you don't have the ISP connector backwards - I do that all the time.

cjkeithley, 2019-07-20:
well, I've got it working again, but it wasn't an ISP cable problem, it was *something* amiss with *an* avrdude.conf file (which seem to exist in multiple places).

I barely remember all the things I that tried *didn't* solve the problem (a "avrdude initialization failed, rc=-1" in the IDE error in the IDE), but once I realized there was an issue with the avrdude.conf file, and realized I wasn't going to fix the .conf file problem manually, I did the following:

     deleted the Arduino15 folder

After quitting and restarting the 1.8.9 IDE I successfully [re]built and uploaded the sketch to the T841 board using the USBtinyISP. I made absolutely zero changes to the USBtinyISP, T841 Board, and the ISP cable!   The relaunched IDE created a new/clean arduino15 folder automatically.

So, we're done. Thank you Spence for all your help and for the clarification about using the bootloader.  I'll close this issue.  

ps.
I must be clear to anyone that's following Spence's and my discussion that an avrdude.conf problem is **not** the most likely cause of this sort of issue!  Do **not** simply delete folders, manually muck with the avrdude.conf file, etc.!  It's far, far, more likely that this sort of issue is a problem with the ISP cable or wiring, the reset circuit, etc., etc., so you should fully and completely exhaust those possibilities first.

pps.
Although I couldn't tell you what or how I 'broke' things (by updating the boards using the IDE board manager?) I started out a few hours ago with the IDE and the avrdude command line tool saying they  didn't know about the T841. I then realized the command tool's avrdude.conf file was coming from /usr/cellar so pointed the command line tool at the right .conf file. Then I realized the IDE was pulling its .conf file from inside the Arduino15 folder.  So pointed the command line tool at the same .conf file in Arduino15, and got a "programmer type jtagice3_updi not found" error), with the IDE reporting  "avrdude initialization failed, rc=-1".  I reached the point where the command line avrdude tool *and* IDE found the T841 and could talk with it, but the IDE simply refused to program it (and returned the rc=-1 error). 

What I found as I worked on this is that there are (at least on my dev system) multiple avrdude.conf files (in /usr/cellar, inside the Arduino IDE, in the arduino15 folder, etc.).  I wasn't certain which one should be the default .conf file used by the IDE, and concluded it'd be better to a clean install and start over. Luckily the very first thing I removed and recreated (the Arduino15 folder) resolved the problem.

closed 2019-07-20 by cjkeithley