SpenceKonde/ATTinyCore #329: Missing PCINT on x313 pinout

EmerickH, 2019-07-17:
Just a little correction, ATTiny 2313/4313 has got 3 PCINT registers (1 per Px) with a total of 17 PCINT (on every pin!), the [documentation](https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/extras/ATtiny_x313.md) is only showing 7.
[image]
(I tested them they are working with ATTinyCore)
I could do a PR but I don't know how you are doing your schematics!

I can also write a little guide on how to use PCINT

SpenceKonde, 2019-07-17:
I don't do the diagrams either, nor do I know what is used to create them, they're done by @MCUdude, though he is quite busy now so I'm not sure when he'll have a chance to work on this - apparently we're missing PCINT markings on the x313 diagram. Somehow I thought I'd seen a diagram that didn't show them, but as noted above, they do exist, so we need them shown on the x313 pinout chart.

MCUdude, 2019-07-17:
Huh, according to [this datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-2543-AVR-ATtiny2313_Datasheet.pdf) there are only eight PCINT pins. When looking at a later revision of the schematic there are suddenly 17 PCINT pins. I'll fix it very soon and upload the new pinout here

EmerickH, 2019-07-17:
OK, I found why this difference:
 * The datasheet you linked is only for ATTiny2313 (still in production but older), which has only 8 PCINT
 * Mine is for ATtiny2313**A** (new version) and ATtiny4313, with both 17 PCINT

MCUdude, 2019-07-17:
Good catch! I'll add an asterisk to each PCINT8-PCINT16 labels and a small note below.

SpenceKonde, 2019-07-17:
AAHA! (and there's a PCINT17 too, don't forget that one)

MCUdude, 2019-07-17:
[ATtinyX313.zip](https://github.com/SpenceKonde/ATTinyCore/files/3402932/ATtinyX313.zip)

Yeah, just crop out ATtinyX61. It happens to be on the same page

EmerickH, 2019-07-17:
Ho you did a little mistake: * ATTiny2313A/ATTiny4313 only
After that I promise I don't ask anything else ;)

MCUdude, 2019-07-17:
> Ho you did a little mistake: * ATTiny2313A/ATTiny4313 only

What a mess Atmel made...
Well, right should be right; fixed!

[ATTiny_pinout.zip](https://github.com/SpenceKonde/ATTinyCore/files/3402990/ATTiny_pinout.zip)

SpenceKonde, 2019-07-17:
Thanks for the PR for this! :-)

SpenceKonde, 2019-07-17:
Two notes: 
We don't count reset in the GPIO pin count, because we don't support RSTDSBL with a menu option (you can control the pin, but you have to upload the sketch, and then go in with avrdude and manually set reset disable fuse. This is because almost nobody in arduino-land has an HV programmer that would be needed to reprogram the chip afterwards - so making this easily accessible would set a trap where users unaware of this would brick their ATtiny (this is the same reason I don't have external *CLOCK* (as opposed to crystal) options - people would mistake it for crystal and brick their part in a way that is very difficult to recover from). It is not uncommon for me to see a support inquiry where the user, unsure of what is wrong, has exhaustively tried every combination of menu options :-/

closed 2019-07-17 by SpenceKonde