SpenceKonde/ATTinyCore #331: Compilation fails with arduino-LoRa library due to SPIClass on certain boards

AzureDVBB, 2019-07-18:
Code to reproduce the error
```cpp
#include <LoRa.h>
#include <SPI.h>

void setup() {
  LoRa.setPins(1, 3, 4);

  if (!LoRa.begin(433E6)) {
    while (1);
  }
}

void loop() {
}
```

Error
```
C:\Users\Bence\Documents\Arduino\libraries\LoRa\src/LoRa.h:77:15: error: 'SPIClass' has not been declared

   void setSPI(SPIClass& spi);

               ^

C:\Users\Bence\Documents\Arduino\libraries\LoRa\src/LoRa.h:101:3: error: 'SPIClass' does not name a type

   SPIClass* _spi;

   ^

exit status 1
Error compiling for board ATtiny24/44/84.
```

Tried compiling for all the boards in this core, and it only affects (irrespective of bootloader or not):
- 24/44/84
- 25/45/85
- 261/461/861
- 43
- 1634
- 2313/4313

The rest of the boards compile without error
Note: Using the latest (1.8.9) arduino editor

Raised the issue with the LoRa library and got redirected here, Oiriginal error report:
https://github.com/sandeepmistry/arduino-LoRa/issues/270

SpenceKonde, 2019-07-18:
Thank you for reporting this serious bug.

SpenceKonde, 2019-07-18:
This will be an easy fix, will submit fix tonight.

commit 2698b3cd (2019-07-18): Fix #331

Should work now!
closed 2019-07-18 by SpenceKonde