SpenceKonde/ATTinyCore #334: Reported sketch size bloat vs damellis core

SpenceKonde, 2019-07-18:
Post #6 here
https://forum.arduino.cc/index.php?topic=613074

This should not be happening. From whence cometh the bloat?

SpenceKonde, 2019-07-24:
I have reproduced this issue, it can be observed with simple sketches like blink.

SpenceKonde, 2019-07-24:
Made some improvements here for digitalWrite and analogWrite - though I don't think this accounts for the trouble that was reported; need example sketch to find that.

sleemanj, 2019-07-24:
Since the damells' core is not very capable (calling it a core is rather a stretch!), I assume it does not support tone out of the box on the 85, and that ATTinyCore does, that is probably the cause of the observed bloatyness because the secondary timers are still setup.

I don't recall if your stock ATTinyCore has a NO_TONE or similar define, if so try a blink sketch with and without it, see how much difference it makes.

SpenceKonde, 2019-07-24:
In the case of blink with added analogWrite() (ie, digitalwrite high, 1s delay, analogwrite 128, 1s delay, digital write low, 1s delay) there were 28 bytes of bloat vs damellis core, due to unnecessary calls to cbi(TCCRnA,COMnx0) when that bit is never set by the core (and if you set it yourself, you've voided the warranty that the builtin functions will do what you want). I reduced the size of that test sketch by 46 bytes, so it's now 18 bytes smaller. His core is less efficient in a number of other ways, so the secondary timer setup winds up not being enough to push my sketch sizes above his (the overhead is pretty small here)

sleemanj, 2019-07-24:
Out of interest what does your compiled blink come to?

sleemanj, 2019-07-24:
And what does his?

SpenceKonde, 2019-07-24:
was 798 vs 770, now 752.

sleemanj, 2019-07-24:
Interesting they are so close.

FWIW, in my fork, standard blink except for adding `#define LED_BUILTIN 2`, 

  * Everything enabled 672 bytes compiled size
  * Tone disabled 210 bytes compiled size
  * Tone and millis both disabled 98 bytes compiled size  (`delay()` does not require `millis()`)

But my tone and your tone will be different, and I rewrote the millis/micros/delay implementation, so apples and oranges really, but providing a tools option to disable the `tone()` initialisation I can imagine might still be helpful to your users :-)

SpenceKonde, 2019-07-24:
210 bytes without tone? Holy shit man... That seems really suspicious - are you sure you're not breaking more than just tone? 
If your fork has the assembler listing generation feature, can you do export compiled binary on those and send me the .lst with everything enabled, and tone disabled? That's just an insane amount of space savings, and having just spent an hour and change staring at dissembler output for blink, I can't see how that's possible without taking out the mechanics behind millis() as well.

sleemanj, 2019-07-24:
As I say, I rewrote millis/micros/delay (not to mention optimisations for constant pin mode/read/write which will also be in play here), so it won't be comparable really for you as mine probably optimises out more than you can, but I've attached disassembly, objdumps of the non-LTO (216 bytes) and LTO (210 bytes), elf's also attached.

[lto.txt](https://github.com/SpenceKonde/ATTinyCore/files/3424928/lto.txt)
[non-lto.txt](https://github.com/SpenceKonde/ATTinyCore/files/3424929/non-lto.txt)

[elfs.zip](https://github.com/SpenceKonde/ATTinyCore/files/3424938/elfs.zip)

[lto-objdump.txt](https://github.com/SpenceKonde/ATTinyCore/files/3424952/lto-objdump.txt)
[nolto-objdump.txt](https://github.com/SpenceKonde/ATTinyCore/files/3424953/nolto-objdump.txt)

As always, welcome to grab anything you want from [my repo](https://github.com/sleemanj/ATTinyCore) note that the fork may or may not work in it's own right, I do the [distribution here](https://github.com/sleemanj/optiboot/blob/master/dists/README.md) which includes other packaging stuff to bring it all together.

sleemanj, 2019-07-24:
A better comparison is perhaps "BlinkWithouthDelay", since that includes millis use.  

  Tone disabled: 330 bytes
  Tone enabled: 668 bytes

And yes, I did test it on an 85 and both work ;-)

The tone initialisation really does take up quite  a bit of space in my fork at least, maybe yours is different.

SpenceKonde, 2019-08-17:
Closing, as this is addressed by changed in 1.2.5. Upcoming 1.3.0 release's tinyNeoPixel library will let us beat other cores for neopixel flash use as well.

closed 2019-08-17 by SpenceKonde