SpenceKonde/ATTinyCore #335: ATTiny1634 QFN run only at 1MHZ

mariusmotea, 2019-07-25:
Hi,

I flash the basic led blinking sketch with one led connected to `PIN_A6` with clock at 8Mhz/internal and i notice that instead of 1 second interval on led change there where 8 seconds. I test then with clock at 12Mhz/external (crystal connected) and i saw the change interval increased to 12seconds. With the clock set at 1Mhz/internal the blinking interval is the correct one, every second. I test with Arduino 1.8.8 and 1.8.9 on MacOS. I want to mention that i move the chip from another board with the heat gun, it is possible to have something broken in the microcontroller caused by the heat?

Board settings:

Board: "ATTiny1634 (no bootloader)"
Clock: "8MHZ (internal)"/ "1MHZ (internal)"/"12MHZ (external)"
B.O.D. Level: B.O.D. enabled 1.8V
B.O.D. Mode (active): B.O.D. Disabled
B.O.D. Mode (sleep): B.O.D. Disabled
Save EEPROM: EEPROM retained
LTO: Disabled

Sketch:
```
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(PIN_A6, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(PIN_A6, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(PIN_A6, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```

SpenceKonde, 2019-07-25:
Did you do "burn bootloader" after choosing the crystal speed? You must
always do this when starting with a fresh chip or changing clock or bod
options, even when a bootloader is in use, as this is the only time that
fuses that set the clock source (as well as the bod settings) are touched.

This behavior mirrors that of the official avr board package and all other
arduino avr board packages.

Be sure to not choose a clock source that uses a crystal when doing "burn
bootloader" if you do not have a crystal connected - this will prevent
further programming, including burning bootloader to set it to use a
different clock source.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

mariusmotea, 2019-07-25:
Thanks @SpenceKonde . I did not burn a bootloader because i believed this is used only to burn new software more easy with serial adapter. I will try to burn the bootloader with 8Mhz internal clock, if this is enough i will not need to use an external crystal.

SpenceKonde, 2019-07-25:
Yes - if you don't wish to upload via serial, you can use the (no bootloader) board definition. 

I wish there was a way to change what the menu option for "burn bootloader" said - maybe "set fuses" for boards without a bootloader, but the IDE does not provide this level of customization to third party hardware packages.

closed 2019-07-25 by SpenceKonde