SpenceKonde/ATTinyCore #336: ATtiny1634 PB3 not available for IO

hseibel12, 2019-07-25:
I was testing a board where I used PB3 as an output. It did not work. Then I tried to configure the Pin as input w/wo pullup. It also did not work. When playing directly with DDRB, PORTB and PUEB nothing happend. Also pwm does not work. Tried another board - same issue. Is this a limitation or a bug?. Core is ATtinyCore 1634 with bootloader.

SpenceKonde, 2019-07-25:
Hi

I saw your review on Tindie. That pin should absolutely work, I was planning to investigate last night and reply at that time, but wound up not having time to test yesterday. I will test today after work. I don't see any obvious problems in the code that would explain why that pin isn't working for you.

SpenceKonde, 2019-07-26:
I cannot reproduce this issue - this sketch works without issue. Do you confirm that this sketch does not work for you?

`
//#define LED_BUILTIN PIN_B3  //this also works
#define LED_BUILTIN 14
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  analogWrite(LED_BUILTIN,128);
  delay(1000);
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
`

Assuming that you still see the failure with that test sketch, that would imply a hardware problem. 
How is the pin behaving? Is it acting as if stuck as OUTPUT with either HIGH or LOW, or tristated/high impedance? 

If it's tristated/high impedance - does the solder joint on that pin look okay? It's been a long time since I had a pin fail to solder in the reflow oven. 

You mentioned in the Tindie review that you had been using it for several projects - at any time might any of the following scenarios have occurred? 
* Pin connected to an external voltage while chip was not powered
* Pin exposed to an external voltage higher than Vcc or lower than Gnd
* Pin set as output connected to a load that tries to draw more than 20mA
Any of these can "blow" a pin, typically resulting in it "stuck" as an OUTPUT with either HIGH or LOW (depending on which driver was damaged), or (if both were damaged), stuck outputting an intermediate voltage (because both "stuck on" drivers are fighting eachother and the voltage ends up in the middle).

hseibel12, 2019-07-26:
I have tested your sketch and you are right: It works if you use this pin as an output!
But, please test the following sketch. I am doing the same for PA0 and for PB3, and I found a difference. While PA0 toggles with the state of its output, PB3 does not:
      --> PB3(HI)=0   PA0(HI)=1
      --> PB3(LO)=0   PA0(LO)=0

Also, if I have PB3 configured as an input + pullup, it does not read 1.

Seeams that it is not possible to read the output state. I have testet on 2 different boards. 

I used this sketch:

#define LED_BUILTIN PIN_B3 //this also works
//#define LED_BUILTIN 14
#define TEST_PORT 8
int val;
void setup() {
pinMode(LED_BUILTIN, OUTPUT);
pinMode(TEST_PORT, OUTPUT);
Serial.begin(9600);
delay(500);
}
void loop() {
digitalWrite(LED_BUILTIN, HIGH); // turn the LED on (HIGH is the voltage level)
digitalWrite(TEST_PORT, HIGH); // turn the LED on (HIGH is the voltage level)
  val = digitalRead(LED_BUILTIN);
  Serial.print("PB3(HI)=");
  Serial.print(val);
  Serial.print("   ");
  val = digitalRead(TEST_PORT);
  Serial.print("PA0(HI)=");
  Serial.println(val);
delay(1000); // wait for a second
analogWrite(LED_BUILTIN,128);
delay(1000);
digitalWrite(LED_BUILTIN, LOW); // turn the LED off by making the voltage LOW
digitalWrite(TEST_PORT, LOW); // turn the LED off by making the voltage LOW
  val = digitalRead(LED_BUILTIN);
  Serial.print("PB3(LO)=");
  Serial.print(val);
  Serial.print("   ");
  val = digitalRead(TEST_PORT);
  Serial.print("PA0(LO)=");
  Serial.println(val);
delay(1000); // wait for a second
}

hseibel12, 2019-07-27:
Hello Spence,

 

thank your very much for your reply! I have tested the sketch and can partially confirm that you were right. I checked again and found:

 

                the PIN can be set to output and works well

but, the PIN cannot be set as an input

 

I have added my sketch on github. Maybe you can check this again? I am not sure if this happens if a port has been damaged (I tested on two boards with same results).

 

Regards

Horst

SpenceKonde, 2019-07-29:
Thanks for your help in investigating this. 

Turns out, the answer was in the datasheet all along, in the errata section. Turns out there is a design flaw in the chip: 

> • **Port Pin Should Not Be Used As Input When ULP Oscillator Is Disabled** Port pin PB3 is not guaranteed to perform as a reliable input when the Ultra Low Power (ULP) oscillator is not running. In addition, the pin is pulled down internally when ULP oscillator is disabled.
> Problem Fix / Workaround: The ULP oscillator is automatically activated when required. To use PB3 as an input, activate the watchdog timer. The watchdog timer automatically enables the ULP oscillator.

I will get a note added to the pinout diagram distributed with the core.

closed 2019-07-29 by SpenceKonde