SpenceKonde/ATTinyCore #337: Need updated pinout diagram for tiny1634

SpenceKonde, 2019-07-29:
@MCUdude 

After a customer complaint and considerable investigation (#336) it was realized that there is a relevant silicon errata regarding PB3 (arduino pin 14) on the ATtiny1634 - it doesn't work as an input (for analog or digital read) except when the watchdog timer is running. This was conspicuously noted in the datasheet, in the errata section on page 283 of the datasheet (can't imagine how I missed it *eyeroll* )

We need an added note for this pin. My thought is a box in a different color next to that pin reading NOTE

and then below the diagram:
"Note: Pin 14 (PB3) does not function as an input (analog or digital) unless the watchdog timer is running (see Errata section of the datasheet for more information)"

MCUdude, 2019-07-30:
Wouldn't it be better to add the note to the ATtiny_1634.md file as text, rather than "burning" it into the pinout diagram? This isn't a chip feature, it's an error that affects a chip revision. I assume users read the ATtiny_1634.md file before designing a product around it?  I would at least if I was using ATTinyCore.

SpenceKonde, 2019-08-02:
Well, yes, it's specific to certain revisions - but all revisions that Atmel/Microchip has shipped are impacted! 
IMO it is grossly dishonest that this limitation is only mentioned on page 283 of the goddamned datasheet, so people could easily design boards and have them fabricated before discovering that PB3 doesn't work, and then Atmel could say "well, we told you so at the very end of a 300-page document, didn't you read it?"

People are pretty much guaranteed to look at the pinout diagram when selecting pins, and there are places where the pinout diagram is presented without the rest of the .md file (for example, on my Tindie store, and on the main page of the ATTinyCore documentation). Since this effects all 1634 chips, and has a significant impact on pin selection, I would tend to favor it being on the chart that people are sure to be looking at while selecting which pins they will use for what.

MCUdude, 2019-08-02:
So _all_ chip revision to this day is affected? That's terrible. Even more terrible is that this is hidden so deeply in the datasheet. I'll add it to the main pinout diagram later today!

SpenceKonde, 2019-08-02:
Yup! As far as I can tell, there is no Rev. D! Not that there's really a
way to tell which silicon rev. you have...

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

MCUdude, 2019-08-03:
Is this pinout diagram OK?
[ATTiny1634.png.zip](https://github.com/SpenceKonde/ATTinyCore/files/3464386/ATTiny1634.png.zip)

SpenceKonde, 2019-08-04:
Great! Thanks!

closed 2019-08-04 by SpenceKonde