SpenceKonde/ATTinyCore #338: Software I2C Master doesn't work on ATTINY828

MatCat, 2019-07-30:
I tried to use I2C on ATTINY828, doesn't work, seems that a function is called deep in the library that is disabled by a #define.

SpenceKonde, 2019-07-31:
Can you let me know what error you're getting? And the code that generates it?

Also - when you say Software I2C Master - are you referring to the builtin one via Wire.h? That is what you should be using - with ATTinyCore (at least recent versions of it) there is no need to use the SoftwareI2CMaster library

MatCat, 2019-07-31:
No error, just hangs the MCU.  I have tried the built in Wire,h, I tried TinyWireM, and I tried SoftI2CMaster, none of them work,

MatCat, 2019-07-31:
ERr sorry, hangs the MCU with the other libraries, with the built in wire it complains about a missing function,  C:\Users\MatCat\AppData\Local\Temp\ccgDQwDd.ltrans0.ltrans.o: In function `main':
<artificial>:(.text.startup+0x2c8): undefined reference to `TwoWire::begin()'

SpenceKonde, 2019-07-31:
Hm, I can compile I2C master code using builtin Wire.h without issue. Can you post the code that's giving that error?

MatCat, 2019-07-31:
It's got 2000 lines of code... but I did create a new blank project with just this, same issue:

#include <Arduino.h>
#include <Wire.h>

void setup() {
  // put your setup code here, to run once:
  Wire.begin();
}

void loop() {
  // put your main code here, to run repeatedly:
}

MatCat, 2019-07-31:
I just did some research,  I don't think PlatformIO uses your library, I thought it did.

SpenceKonde, 2019-07-31:
My guess is it *does* use my board package, just a very old version of it.....

SpenceKonde, 2019-08-08:
Closed, since not a problem with the ATTinyCore in it's current incarnation, but with what is in platformIO.

closed 2019-08-08 by SpenceKonde