SpenceKonde/ATTinyCore #339: [FYI] Installing Arduino megaAVR Boards breaks ATTinyCore ISP sketch uploads

JChristensen, 2019-08-07:
I'm not sure where the difficulty lies; I'm opening an issue here too just to ensure you're aware and point to the fully documented issue:
https://github.com/arduino/ArduinoCore-megaavr/issues/47

SpenceKonde, 2019-08-07:
Yikes! This is nasty. I will need to investigate this. I'm 99% sure USBAsp is not impacted by this. I wonder why USBTinyISP is?

JChristensen, 2019-08-07:
I tested with a USBasp programmer, you are correct, there is no issue. However, avrdude still runs from the other location; that's the part that seems weird to me.

SpenceKonde, 2019-08-07:
I wonder if there's a way to tell it to use the old version of avrdude? 
@per1234 - do you know if there's a way to do that, or have any other thoughts on this?

per1234, 2019-08-07:
This problem with avrdude 6.3.0-arduino16 has been investigated and explained here: https://github.com/arduino/ArduinoCore-avr/issues/91

>I wonder if there's a way to tell it to use the old version of avrdude?

Yes, it only requires defining the toolsDependencies in the Boards Manager JSON file like we did for megaTinyCore:
https://github.com/SpenceKonde/ReleaseScripts/pull/21

That only does it for Boards Manager installations. For manual installations, I believe we can specify specific tool versions in platform.txt like `runtime.tools.avrdude-6.3.0-arduino14.path` but that will only be defined if the user has that specific version of avrdude installed already. If not, it will break uploads.

SpenceKonde, 2019-08-07:
Aaaaha! Thanks for the issue that tied it all together! 

Yeah, let's make that so for the board manager version.... 

Not sure what to do about the manual installations yet - maybe just a caveat in the documentation and hopefully the official megaavr package will be sorted out soon and we won't have to worry so much. 

Is there an avrdude package that we can reference that has the latest fix discussed in that issue? If so, let's make sure toolsDependencies for megaTinyCore installs that one, so at least megaTinyCore doesn't trigger the issue for this core.

per1234, 2019-08-07:
>Is there an avrdude package that we can reference that has the latest fix discussed in that issue?

Yes. avrdude 6.3.0-arduino17 [is claimed to have the bug fixed](https://github.com/arduino/ArduinoCore-avr/issues/91#issuecomment-502138990) and is available. The reason I didn't use that in the megaTinyCore `toolsDependencies` is because I thought it was safest to use the tool versions used by the latest Arduino megaAVR Boards release. However, a lot of people are using an Arduino IDE test build that bundles avrdude 6.3.0-arduino17 for Arduino AVR Boards due to that test build providing support for macOS Catalina and there have been no reports of problems with avrdude 6.3.0-arduino17 so I think it's pretty safe. @MCUdude did you ever get a chance to test out avrdude 6.3.0-arduino17?

>Yeah, let's make that so for the board manager version....

OK. Should I use avrdude 6.3.0-arduino14 (the version used by Arduino AVR Boards 1.6.23) or avrdude 6.3.0-arduino17? There is no avrdude 6.3.0-arduino15 available.

SpenceKonde, 2019-08-08:
Let's go with arduino17 for both.

per1234, 2019-08-08:
It turns out I can't retroactively set the avrdude `toolDependencies` version to `6.3.0-arduino17` in the Boards Manager JSON file entries for the existing ATTinyCore releases (and megaTinyCore too, though there is an additional problem blocking doing it for that core as well: https://github.com/SpenceKonde/megaTinyCore/pull/73). Although that would be fine for new installations of ATTinyCore, for existing installations, the Arduino IDE constantly downloads fresh copies of http://drazzy.com/package_drazzy.com_index.json and uses the tool versions specified in the entry for the currently installed core version. avrdude 6.3.0-arduino17 will almost certainly not happen to be installed for the users with existing ATTinyCore installations and the missing tool version would cause uploads to fail. So it will need to wait to be done when I update the JSON file at the next release of ATTinyCore (and only done for that release and subsequent releases).

MCUdude, 2019-08-08:
@per1234 I have tested 6.3.0-arduino17 and can confirm that it works with USBtinyISP. USBasp was never affected AFAIK.

SpenceKonde, 2019-08-14:
https://github.com/SpenceKonde/ReleaseScripts/issues/25 - this is a change that goes into the board manager json, which is over in that repo. Closing this so these issues are tracking changes that happen in this core only.

closed 2019-08-14 by SpenceKonde