SpenceKonde/ATTinyCore #340: TwoWire::available needs a return value for WIRE_SLAVE_ONLY

PaulZC, 2019-08-14:
Hi Spence,
Sincere thanks for sharing this work - much appreciated!
TwoWire::available is missing a return value for the WIRE_SLAVE_ONLY case:
See [these lines](https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/libraries/Wire/src/Wire.cpp#L870-L871)
All the best,
Paul

PaulZC, 2019-08-14:
Could this be the root cause of mystery issue #229 ?!
:grinning:

SpenceKonde, 2019-08-14:
Wow. Thanks for doing the work to look into this - I have to admit, I2C slave mode has seen only cursory investigation by me.

closed 2019-08-14 by SpenceKonde

SpenceKonde, 2019-08-14:
Sadly, this doesn't address those other two issues, as this applies to the 841/1634/828, while those apply to USI-based ones.

jseaber, 2022-01-06:
Sorry to jump into this so late, and my apologies in advance for vagueness. I've confirmed an I2C Slave bug on ATTiny88 and have a workaround in place. I'm fairly confident this is related, but please correct me if I'm wrong:

## Hardware

- I2C master is an XU208 (FPGA) running [lib_i2c](https://github.com/xmos/lib_i2c) at 400kHz with 10k pullups to 3.3VDC:
- ATTiny88 is configured as below, running ATTinyCore v1.5.2 
- Two other devices are configured as I2C slaves, which I cannot publicly list at this time

```
void initialize_i2c_slave(void)
{
  Wire.setClock(400000);
  Wire.onReceive(receive_i2c_event);
  Wire.onRequest(request_i2c_event);
  Wire.begin(PROTOCOL_ADDR);
}
```

## Behavior

I've observed a rare boot failure of the Master upon warm resets while the ATTiny88 slave remains powered.

We managed to capture I2C via logic analyzer during the glitch, and found the Master was successfully writing to all I2C slaves _except_ for the ATTiny88, which resulted in 100% NAKs. It's possible there is an I2C signal violation, but we have yet to pinpoint the exact reason. At any rate, the ATTiny88 remains unresponsive via I2C until either the ATTiny88 is reset, or TWI is reset.

## Workaround

For now, we call the function below when the bug arises (credit due to [this guy](https://forum.arduino.cc/t/looking-for-a-way-to-reset-arduino-i2c-hardware-arduino-as-i2c-slave/587482/6)):
```
void twi_restart(void)
{
  // disable twi module, acks, and twi interrupt
  TWCR &= ~(_BV(TWEN) | _BV(TWIE) | _BV(TWEA));

  // deactivate internal pullups for twi.
  digitalWrite(SDA, 0);
  digitalWrite(SCL, 0);

  delay(5);

  // activate internal pullups for twi.
  digitalWrite(SDA, 1);
  digitalWrite(SCL, 1);

  // initialize twi prescaler and bit rate
  cbi(TWSR, TWPS0);
  cbi(TWSR, TWPS1);
}
```
  
It's possible one of the slaves or master is violating I2C protocol, causing the ATTiny88 to raise a TWI Status Code and halt I2C transmission. See Tables 15-4 and 15-5 in the ATTiny88 datasheet.

I can provide Salae Logic dumps upon request, and will be glad to test further. Hope this helps!

commit 0d17000a (2019-08-14): Correct missing return value; statement (fix #340)