SpenceKonde/ATTinyCore #341: MacOS Catalina requires 64bit executables

JohnOH, 2019-08-15:
I loaded up this application using board manager on Arduino 1.8.10 running on the latest beta of Catalina. I had already been required to load a 64bit version of Arduino  to build sketches for ATMega328. When I attempt to build ATTiny84 sketches I see:
> fork/exec /Users/johare/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++: bad CPU type in executable

I am guessing I need to update 4.8.1-arduino5/bin/avr-g++ to a 64bit version. Any thoughts on my problem will be welcome.

SpenceKonde, 2019-08-15:
Are you able to compile code for normal Arduino boards? I suspect you will get the same issue there.

JohnOH, 2019-08-16:
When I initially installed Catalina I saw the problem as you suggest. I found instructions to update Arduino to a 64bit version which I did. Since then I have been able to build sketches for 328P and 2560 devices. I used board manager to install the ATTinyCore and only see the problem when selecting to build for ATTiny devices. 

Does ATTinyCore use a specific g++ or does it use the same executable as the 328p & 2560?

JohnOH, 2019-08-16:
When I build for 2560 a different version ofg++ appears to be used:
> /Users/johare/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-I/private/var/folders/6c/gxr4fz9s0_1cfcfnq7b8fmr40000gp/T/AppTranslocation/80E02753-9109-4164-BABB-8BB171E4F26C/d/Arduino 5.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/private/var/folders/6c/gxr4fz9s0_1cfcfnq7b8fmr40000gp/T/AppTranslocation/80E02753-9109-4164-BABB-8BB171E4F26C/d/Arduino 5.app/Contents/Java/hardware/arduino/avr/variants/mega" /var/folders/6c/gxr4fz9s0_1cfcfnq7b8fmr40000gp/T/arduino_build_897183/sketch/RFxConsole.ino.cpp -o /dev/null

SpenceKonde, 2019-08-16:
@per1234 - can we also tell board manager for the next version to use 7.3.0 compiler version when it's installed via board manager? (for non-board-manager installations, ideally I'd just like it to stick to using the latest version of avrgcc that it can find, which I think is the current board manager behavior).

JohnOH, 2019-08-17:
I look forward to trying it out with 7.3.0 compiler.

per1234, 2019-08-17:
>can we also tell board manager for the next version to use 7.3.0 compiler version when it's installed via board manager?

Yes. Will do.

SpenceKonde, 2019-08-17:
As a stop-gap solution, if you remove the core via board manager and do a manual installation, and some other board package has installed 7.3.0, it should use that. 
But I'm hoping to release the next version by the end of the weekend to get this change, tinyNeoPixel, and a few other changes out.

SpenceKonde, 2019-08-17:
This change goes into the board manager json, not the core proper, so closing this one, see https://github.com/SpenceKonde/ReleaseScripts/issues/25

closed 2019-08-17 by SpenceKonde