SpenceKonde/ATTinyCore #342: Works with 45 but not 85

jtommi, 2019-08-15:
ATTiny85s give me an invalid device signature error.
The signature is 0x000000
I tried with 3 chips.
But if I take it out, put an ATTiny45 in place and changing "chip" to the correct chip, it works, tested with 2 ATTiny45.
Do you have any explanation for this or are all 3 ATTiny85 broken ?

sleemanj, 2019-08-15:
Indicates that programmer can not talk to the 85.  Since you have eliminated wiring issue by showing you can reliably communicate with a 45 with the exact same setup, then that leaves the 85 as the most likely problem.  

The first thing to do is try a 45 but select 85 as the chip in the IDE, of course it will fail but it should give you the 45 signature instead of 0x00000000.

Assuming it gives you different numbers then that adds weight to the problem being in the 85 chips themselves.  Rather than broken they are probably configured such that you can not program them most likely reset disabled, SPI disabled, setup for crystal with none present, setup for clock with none present.

SpenceKonde, 2019-08-15:
Are these attiny85's pulled from those digispark-clone boards that are programmed directly over USB? Some of those get programmed with reset disabled; Only an HV Serial Programmer can reenable reset so that they can be programmed normally.

jtommi, 2019-08-16:
Wow, thanks for the fast help.

> The first thing to do is try a 45 but select 85 as the chip in the IDE, of course it will fail but it should give you the 45 signature instead of 0x00000000.

I already did that before, it said indeed wrong signature and that the signature looks like a 45T (don't remember the letter exactly).

The chips were separately purchased from Aliexpress, so not the most reliable source. The 45s might be from Futurlec

SpenceKonde, 2019-08-16:
I would be suspicious that the chips may have been pre-loaded with digispark firmware... That seems to be super popular lately, such that many people think that's the "normal" way to upload to t85's. They also could have been damaged in transit - it's yet to happen to me, but packaging used by chinese shippers is, as you may have noticed, not quite as rigorous compared to respectable domestic vendors.

SpenceKonde, 2019-08-17:
Closing because not an issue with the core.

closed 2019-08-17 by SpenceKonde