SpenceKonde/ATTinyCore #346: i can't compile the examples with attiny 261/461/861

Master811129, 2019-08-23:
http://uupload.ir/files/o4il_bug.png

hello i have a problem with the latest version of this core i cannot compile ANY sketch in my 1.8.9 Arduino IDE.
Even the core example and BLINK sketch i think this is a bug.
thank you.

SpenceKonde, 2019-08-23:
Gaaaaaaah!

I'll do a 1.3.1 release today to fix this regression - I just forgot to
change two lines of boards.txt, and because travis needs to be updated to
account for the new menu options, this wasnt flagged internaly

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2019-08-23:
Note that this only effects the x61 parts.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2019-08-23:
This is fixed now. Fix in in 1.3.1, which will be available in board manager soon after @per1234 updates the board manager json files. Fix is already available in github version and 1.3.1 manual installation package.

closed 2019-08-23 by SpenceKonde

Master811129, 2019-08-23:
> This is fixed now. Fix in in 1.3.1, which will be available in board manager soon after @per1234 updates the board manager json files. Fix is already available in github version and 1.3.1 manual installation package.

so i'm waiting for this update to finish my project.
thank you for your quick responsibility.

SpenceKonde, 2019-08-23:
In the meantime you can work around it by choosing Timer1 clock source as something other than CPU (default). This will change the frequency of PWM output (increasing it) but will not otherwise change behavior.

commit e45a21d7 (2019-08-24): fix #346 in x61 optiboot