SpenceKonde/ATTinyCore #351: SPI not working in ATtiny861

ArjFPGA, 2019-09-11:
To test the SPI in attiny 861 I uploaded a master transmit sketch to attiny(via arduino(uno) as ISP) and slave recieve sketch to another Arduino uno.
 The  Arduino(slave) should output  "Hello world!" to the serial monitor, which it isn't doing. I doing quite understand where the problem is, hardware or software.
I think the hardware is working fine(it is a custom designed PCB).Blink sketch is working and digitalRead() function is also working fine. If it is software where the issue is?Is it in the code or core.

Here is the code.
Master transmit.This is the code which is uploaded into the attiny 861.
`
#include <SPI.h>

void setup (void) {
    //set baud rate to 115200 for usart
   pinMode(14, OUTPUT);
   pinMode(9, OUTPUT);
   pinMode(1, OUTPUT);
   digitalWrite(14, HIGH); // disable Slave Select
   SPI.begin ();
   SPI.setClockDivider(SPI_CLOCK_DIV8);//divide the clock by 8
}
 
void loop (void) {
   char c;
   digitalWrite(14, LOW); // enable Slave Select
   // send test string
   for (const char * p = "Hello, world!\r" ; c = *p; p++) 
   {
      SPI.transfer (c);

   }
   
 
   digitalWrite(14, HIGH); // disable Slave Select
   delay(2000);
}`

This is the slave recieve code.

`#include <SPI.h>
char buff [50];
volatile byte indx;
volatile boolean process;

void setup (void) {
   Serial.begin (115200);
   pinMode(MISO, OUTPUT); // have to send on master in so it set as output
   SPCR |= _BV(SPE); // turn on SPI in slave mode
   indx = 0; // buffer empty
   process = false;
   SPI.attachInterrupt(); // turn on interrupt
}

ISR (SPI_STC_vect) // SPI interrupt routine 
{ 
   byte c = SPDR; // read byte from SPI Data Register
   if (indx < sizeof buff) {
      buff [indx++] = c; // save data in the next index in the array buff
      if (c == '\r') //check for the end of the word
      process = true;
   }
}

void loop (void) {
   if (process) {
      process = false; //reset the process
      Serial.println (buff); //print the array on serial monitor
      indx= 0; //reset button to zero
   }
}`

Thanks in advance.

SpenceKonde, 2019-09-11:
Does this work with a second Arduino Uno (or other atmega-based board) as the master?

The first thing that jumps out at me is that buff is not declared volatile on the slave, which could let the compiler optimize it out... But SPI on the parts with a USI (like the '861) has not been well exercised; particularly the 861, which seems to get very little attention from users.

ArjFPGA, 2019-09-12:
Yes, this works with Arduino uno.

SpenceKonde, 2019-09-15:
Do you have the MISO/MOSI pins the right way around? 
Per docs:
> Take care to note that the USI does not have MISO/MOSI, it has DI/DO; when operating in master mode, DI is MISO, and DO is MOSI. When operating in slave mode, DI is MOSI and DO is MISO. 

The pins marked in dark blue are for ISP programming, the ones in teal are for using SPI from the sketch. 

I see you setting the DI pin as OUTPUT - while the SPI library will undo this mistake for you, it suggests that you think that this pin should be MOSI, rather than MISO, and hence may have connected DI as if it were MOSI, and DO as if it were MISO. This is backwards.

SpenceKonde, 2019-09-23:
Can you comment on this before I start an independent investigation? (ie, just swap the MISO/MOSI wires, and see if that fixes it)

ArjFPGA, 2019-11-15:
Sorry.I was busy for sometime and forgot about this.I will test what you said as soon as possible.

ArjFPGA, 2019-11-15:
I am programming the ATtiny861 using Arduino uno. Using SPI.Where do i need to connect my SPI device to? MOSI&MISO or DI&DO?

ArjFPGA, 2019-11-15:
Thanks for your replies

SpenceKonde, 2019-11-15:
When programming via SPI/ISP, you use the pins marked on the chart as MISO/MOSI (note that these are marked as "programming pins" in the pinout chart).

When using the t861 as a master, the DI pin is used as MISO, and the DO pin is used as MOSI. When using the t861 as slave, the DI pin is used as MOSI, and the DO pin is used as MISO. 

Hope this helps.

ArjFPGA, 2019-11-17:
Hi .Can you please send a code for testing SPI with t861 as master transmitter and arduino uno as slave reciever?

SpenceKonde, 2019-12-12:
Did this get it working, or are you still having issues with it?

SpenceKonde, 2019-12-14:
Closing and assuming that above advise solved problem. If it did not, please reopen and we can continue investigation.

closed 2019-12-14 by SpenceKonde