SpenceKonde/ATTinyCore #353: ATTiny85 - pin 0 not working

JRAlvarez2017, 2019-09-24:
Programming the ATTiny85 pin 0 as OUTPUT does not work.
pinMode(0, OUTPUT);
digitalWrite(0, HIGH); or dogitalWrite(0, LOW)
and the pin shows 2.5 volts.

sleemanj, 2019-09-24:
Have you disconnected your programmer and/or serial comms device when conducting the measurement?  This pin is MOSI for programming, and TX for typical serial comms.

Also are you sure you are measuring Digital Pin 0, that is the bottom right hand pin, not the top right hand pin (Pin 1, Reset).

SpenceKonde, 2019-09-24:
You're not the same person who posted on arduino forums asking how to increase output voltage on pin 1 are you?

SpenceKonde, 2019-09-24:
This does not fail for me.

Does this happen with another chip? Are you measuring the correct pin? 

~2.5v output on a pin, regardless of what you tell it to do, can be a symptom of a blown pin - was the pin used to drive an excessive load or exposed to external voltage higher than vcc, lower than ground, or while the chip was not powered?

Jose2059, 2019-09-24:
That is what I thought, a pin with the output damaged. Then I tried a new pin, and the same, so I thought it is something else. Finally, I used a different library ATTiny, and it works as expected.
I am using visual micro and visual studio. Did not tried with arduino IDE.

JRAlvarez2017, 2019-09-26:
I found what is causing the issue. I was using the library SimpleTime.h in this sketch.

SpenceKonde, 2019-09-26:
Glad to hear problem found. Does the behavior differ from other colors, or is this a problem with that lubrary?

JRAlvarez2017, 2019-09-26:
Well, when I use a pure sketch setting pin 0 as OUTPUT, and HIGH, it works (after removing the SimpleTime.h library). However, when I use millis() to do the timers, it fails again. 
It seems something in millis() is using pin 0.

JRAlvarez2017, 2019-09-26:
What do you mean with other colors?
.."Does the behavior differ from other colors.."

JRAlvarez2017, 2019-09-26:
I tried with the Arduino IDE, and  I have the same issue.

SpenceKonde, 2019-10-07:
Can you try putting this in setup() and see if this fixes it? 
TCCR0A=(TCCR0A&0x03);

If it does, I know what the issue is - I will eventually take a look at this myself if I don't hear back, but if you can test that, and it fixes it, that will confirm that it's a bug and what the bug is and I can get it fixed quickly.

SpenceKonde, 2019-10-08:
I have confirmed that I cannot reproduce this issue. 

Can you post a sketch that demonstrates the issue, and where you got this SimpleTime.h library, which seems to cause problems?

JRAlvarez2017, 2019-10-08:
I will do, please give me a few days, kind of busy at work.
Thank you for your replies.

JRAlvarez2017, 2019-10-10:
Spence, I tried today with this simple sketch, and it works fine. I do not know what was the problem before. I will investigate later when I have more time. As I have the original sketch working on several devices, I can double check again. Anyway, whatever it was, it should not be in the ATTinyCore, so you can close the issue.
I appreciate you help, and sorry for the time you spent on it.

unsigned long timeElapsed;

void setup() {
	pinMode(0, OUTPUT);
	digitalWrite(0, LOW);
	pinMode(1, OUTPUT);
	digitalWrite(1, HIGH);
	timeElapsed = millis();
}

void loop() {
	if (millis() - timeElapsed > 500) {
		digitalWrite(0, !digitalRead(0));
		digitalWrite(1, !digitalRead(1));
		timeElapsed = millis();
	}
}

SpenceKonde, 2019-10-13:
You mentioned before that the problem may have been linked to a SimpleTime library? Does that trigger the issue now?

If so, I'd be interested to look at the library and see why, and if it's a necessary consequence of the library, or a bug, and whether the bug is in what that library does, or some non-standard behavior of my core that's causing this issue with a library that should work.

SpenceKonde, 2019-12-13:
Closing issue, feel free to reopen if you have more information as noted above.

closed 2019-12-13 by SpenceKonde