SpenceKonde/ATTinyCore #354: DISABLEMILLIS should leave timer interrupt disabled

tikonen, 2019-09-28:
Hi, 
I'm not very familiar with the code so I might misunderstood something but this looks like a bug.
I'm building for ATTiny2313 and my project crashes after enabling interrupts if I had disabled millis().
Workaround is to define dummy ISR handler 
```
#if defined(DISABLEMILLIS)
/* dummy handler */
SIGNAL(TIMER0_OVF_vect)
{
}
#endif
```
Probably the init function in the ATTinyCore\avr\cores\tiny\wiring.c should leave timer interrupt disabled if the define is set. My guess is that code will fault because the ISR handler is not defined at least on ATTiny2313 target.

Thank you for the great project!

SpenceKonde, 2019-10-03:
Will investigate why the interrupt is getting enabled; it should not be.

SpenceKonde, 2019-10-04:
Thanks for reporting this bug.

commit e8e9e5c6 (2019-10-04): Fix #354 - don't turn on ISR if millis disabled!
commit 66399139 (2019-10-04): Fix #354 - don't enable interrupt if millis disabled
closed 2019-10-04 by SpenceKonde