SpenceKonde/ATTinyCore #356: Failure to upload to ATTiny84

6v6gt-duino, 2019-10-04:
**Summary:**
ATTinyCore version 1.3.2
Arduino IDE 1.8.2
target: ATTiny84

**Error message during upload. The path looks rather odd:**

avr-gcc: error: C:\Users\6v6gt\AppData\Local\Temp\arduino_build_131884/..\arduino_cache_186227\core\core_ATTinyCore_avr_attinyx4_LTO_enable,chip_84,clock_8internal,eesave_aenable,bod_disable,pinmapping_anew,millis_enabled,neopixelport_porta_1bc5d2d7fe299bbd4d4a668366e76c74.a: No such file or directory

**During the prior verify operation, the path is this. Similar but not identical:**

Archiving built core (caching) in: C:\Users\6v6gt\AppData\Local\Temp\arduino_cache_186227\core\core_ATTinyCore_avr_attinyx4_LTO_enable,chip_84,clock_8internal,eesave_aenable,bod_disable,pinmapping_anew,millis_enabled,neopixelport_porta_1bc5d2d7fe299bbd4d4a668366e76c74.a

It works when I revert to ATTinyCore version 1.1.4

Attached file is the complete verify and upload operation

[ATTINY84_Error.txt](https://github.com/SpenceKonde/ATTinyCore/files/3690589/ATTINY84_Error.txt)

SpenceKonde, 2019-10-04:
The root cause of this is an ide bug that was fixed in... 1.8.6 or
something around there. It was storing all the options in the file name of
the cached core, but as more options were added, this wound up being longer
then the windows max filename length. Newer versions of the IDE fix this by
using a hash of the options instead of a list of all of them. Updating to
latest IDE version fixes it.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

6v6gt-duino, 2019-10-04:
OK. That's it. It works fine with IDE version 1.8.9.
I "upgraded" recently to solve another problem and it left appears to have left 2 versions lying around. The IDE start menu opens the newest version. The file associations point to the older version. So, in this particular case, I was using the older IDE.
Many thanks that you so quickly responded. I'll just close this issue.

closed 2019-10-04 by 6v6gt-duino

SpenceKonde, 2019-10-04:
I added a note about this issue in the gotchas section.

6v6gt-duino, 2019-10-04:
Ok. Many thanks for that. I've just looked an seen it. 
Once I get something which works, I'm a bit nervous of upgrading. But from time to time, one has to.
karma++ in the forum
Best regards,
6v6gt