SpenceKonde/ATTinyCore #357: stdlib.h missing

jacky4566, 2019-10-07:
Just downloaded the core and having trouble compiling for ATTINY45.

stdlib.h should be part of the base arduino core no?

Here is my compile error

```C++
In file included from C:\Users\...\arduino_build_373266\sketch\TinyParticleSupervisor.ino.cpp:1:0:

C:\Users\...\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.3.2\cores\tiny/Arduino.h:4:10: fatal error: stdlib.h: No such file or directory

 #include <stdlib.h>

          ^~~~~~~~~~
```

SpenceKonde, 2019-10-07:
I thought that was part of avrlibc, which is supplied with the compiler -
can you compile for an Uno?

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2019-10-07:
Also, what version of the IDE?

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2019-10-08:
Yeah, somethings wrong with the compiler libraries on your system. I couldn't say exactly what though.

SpenceKonde, 2019-10-08:
Does this happen with other sketches? I would expect it to happen with all sketches.

I would suggest asking on the Arduino IDE forums ( forum.arduino.cc ) for advise fixing the problem with your IDE.

jacky4566, 2019-10-08:
Hey Sorry SpenceKonde,

Not sure exactly what the problem was but removing the core and updating to arduino 1.8.10 seems to have fixed it. I suspect I had too many conflicting changes in my arduino install. 

Cheers,

closed 2019-10-08 by jacky4566