SpenceKonde/ATTinyCore #360: Tools -> tinyNeoPixel Port menu is missing

mamert, 2019-10-19:
screenshot: https://photos.app.goo.gl/H3eiq9cAQwzDnz4e6

The tinyNeoPixel menu, mentioned in the readme, is not displayed. Others, like the millis()/micros() menu, for instance, are present.
Tested with Arduino IDE 1.8.8 on both Windows10 and Ubuntu.

Possibly related: the "tinyNeoPixel Static/strandtest" example works, but "tinyNeoPixel/strandtest" does not. I've only changed PIN to 4 and NUM_LEDS to 6.

SpenceKonde, 2019-10-19:
There is no neopixel port menu for the tiny x5 series because there is only
one port on those parts (each port can have up to 8 pins, and there are
only 6 i/o pins counting reset on the x5 ). I will add a note about that to
the documentation.

I'm surprised to hear that the non-static version doesnt work, especially
when the static version does. Will investigate; all the hard stuff is
identical between the two libraries, so its gotta be something silly.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2019-11-18:
Planned version set to 1.3.3, to confirm functionality with non-static tinyNeoPixel library, and correct if bug can be reproduced.

mamert, 2019-11-19:
Thank you.
So it turns out these are 2 unrelated questions.
The main one's been answered
The second one's resolved itself, somehow - I've tried a 3rd Arduino IDE installation, and both library versions work correctly there. With no code differences. I don't know what the reason is, but it's my problem, not yours, thank you for your time :)

closed 2019-11-19 by mamert

SpenceKonde, 2019-11-20:
I blame problems like that on the jumper wires. 

I actually recently discovered that my stockpile of dupont jumper wires has been contaminated with bad ones that look almost identical except that the length of the plastic housing is about 2mm longer, apparently the free jumpers that come with the serial adapters I bundle as an added cost option for bootloader-supporting boards I sell on Tindie don't make a reliable connection! Thankfully, my customers don't need to deal with this, because when I bundle the adapters, I include a (non-defective) 6p jumper cable, instead of the 5x1p jumper wires that come with the adapters. So what I am saying is that "bad jumper cables" are a great and plausible scapegoat.