SpenceKonde/ATTinyCore #362: Missing DDRA x313

DarkElvenAngel, 2019-10-26:
I was trying to squeeze my program down to fit in an ATTiny 2313 only to get all sorts of errors about DDRA and PORTA PIN_A# Not being defined in scope.  I tried to include avr/io.h to no avail.

Am I doing something wrong or is something missing?

Thanks

SpenceKonde, 2019-10-26:
Interesting; I catch this in one direction (if the non-standard DDA mnemonics are not there, I define them as the DDRA ones), but I don't catch it in the other direction. I also didn't realize that the other port a registers have wrong names on the 2313 too. I'll catch these in the variant file in the next release. 
In the meantime, use the standard names, not the weird 2313 names: 
DDAn (not DDRAn), PINAn, PORTAn 
Also - uh.... you do know that all of those are just #defined to n, right (ie, DDRA0 or DDA0 is #defined as 0)? 

PIN_xn is never correct when doing direct port manipulation. No underscore. 

The root cause of this is that atmel fucked up with the 2313 vs 2313a on port A. the 2313 has nonstandard names for the bits in the registers of port A, while the normal names are used for the 2313a. For the 2313, we compile for the a-spec parts, not the 2313 (the code produced is compatible - it's just the names of the registers that are fucked up; the 2313 is also strictly worse than the 2313a, and costs about 50% more, probably because Atmel/Microchip wants people to stop using it and switch to the 2313a so they can stop making the 2313).

DarkElvenAngel, 2019-10-26:
Just my luck! Is there a work around to access the registers ?  I understand how the some the values are just numbers but how about the register names is there some #define for that?

It was expensive when I compared it to other chips online I needed a DIP package and that's was what my local guy had on the shelf I would have rather had the 4313, does it suffer from the same bug?

DarkElvenAngel, 2019-10-26:
I got it working I added a few #define statements

#define PINA _SFR_IO8(0x019)
#define DDRA _SFR_IO8(0x1A)
#define PORTA _SFR_IO8(0x01B)

SpenceKonde, 2019-10-28:
Yeah, they used weird names. The names used in the 2313 datasheet *should* work, but it sounds like they don't. 

I'm going to add the missing #defines to the variant file so it will always have both names defined. 

I'm surprised that PORTA and PINA themselves weren't defined. I'll need to check the io files when I get home to see what it has there (oddly enough google turns up not a single copy of iot2313.h anywhere on the internet!).

SpenceKonde, 2019-12-14:
Hmmmm.... i am looking at the iotn2313.h and iotn2313a.h, and both of them have all of those defined...i also can't reproduce your problem. the only difference i see is the DDRA vs DDA as names for the bits....

SpenceKonde, 2019-12-14:
I am going to close this - if you have a sketch that reproduces this problem if you don't put those defines in, I would love to see it - I will still see emails about this, and if you can provide me with a sketch and the options (board, chip, etc) that reproduces this issue, I would love to see it, and will reopen this issue.

closed 2019-12-14 by SpenceKonde

DarkElvenAngel, 2020-02-23:
Sorry I missed this I have a sketch for you
```
#include <avr/io.h>
#define PINA _SFR_IO8(0x019)
#define PINA0 0
#define PINA1 1
#define PINA2 2

#define DDRA _SFR_IO8(0x01A)
#define DDA0 0
#define DDA1 1
#define DDA2 2

#define PORTA _SFR_IO8(0x01B)
#define PORTA0 0
#define PORTA1 1
#define PORTA2 2

unsigned long previousMillis1 = 0;
unsigned long previousMillis2 = 0;
unsigned long previousMillis3 = 0;
unsigned long previousMillis4 = 0;

const long pulseTime = 500;
unsigned char relayState = 0;

void setup() {
  Serial.begin(57600);
  DDRA |= 0X3;
  DDRD |= 0xC;
}

void loop() {
  unsigned long currentMillis = millis();
  if ((relayState & 0x1) && (currentMillis - previousMillis1 >= pulseTime)) {
    PORTA &= ~(1 << PINA1);
    relayState &= ~0x1;
  }
  if ((relayState & 0x2) && (currentMillis - previousMillis2 >= pulseTime)) {
    PORTA &= ~(1 << PINA0);
    relayState &= ~0x2;
  }
  if ((relayState & 0x4) && (currentMillis - previousMillis3 >= pulseTime)) {
    PORTD &= ~(1 << PD2);
    relayState &= ~0x4;
  }
  if ((relayState & 0x8) && (currentMillis - previousMillis4 >= pulseTime)) {
    PORTD &= ~(1 << PD3);
    relayState &= ~0x8;
  }
  while (Serial.available()){
    char inChar = (char)Serial.read();
    switch(inChar) {
     case '1':
      PORTA |= (1 << PINA1);
      break;
     case '!':
      PORTA &= ~(1 << PINA1);
      break;
     case '2':
      PORTA |= (1 << PINA0);
      break;
     case '@':
      PORTA &= ~(1 << PINA0);
      break;
     case '3':
      PORTD |= (1 << PD2);
      break;
     case '#':
      PORTD &= ~(1 << PD2);
      break;
     case '4':
      PORTD |= (1 << PD3);
      break;
     case '$':
      PORTD &= ~(1 << PD3);
      break;
     case 'Q':
      PORTA |= (1 << PINA1);
      relayState |= 0x1;
      previousMillis1 = currentMillis;
      break;
     case 'W':
      PORTA |= (1 << PINA0);
      relayState |= 0x2;
      previousMillis2 = currentMillis;
      break;
     case 'E':
      PORTD |= (1 << PD2);
      relayState |= 0x4;
      previousMillis3 = currentMillis;
      break;
     case 'R':
      PORTD |= (1 << PD3);
      relayState |= 0x8;
      previousMillis4 = currentMillis;
      break;
     case 'I':
      Serial.write('O');
      break;
    }
  }
}
```
I had to add the define statements so it would work.