SpenceKonde/ATTinyCore #363: Wire library on 841

Yardie-, 2019-10-26:
Hi 
Firstly thanks this is a great library.
I am having a compile error when compiling for an 841
the primary issue seems to be 
.arduino15/packages/ATTinyCore/hardware/avr/1.2.5/libraries/Wire/src/Wire.cpp:912:31: warning: invalid conversion from 'void (*)(int)' to 'void (*)(size_t) {aka void (*)(unsigned int)}' [-fpermissive]

humorously the same section of code in the esp8266 Wire library is 

void TwoWire::onReceive( void (*function)(int) ) {
// arduino api compatibility fixer:
// really hope size parameter will not exceed 2^31 :)
static_assert(sizeof(int) == sizeof(size_t), "something is wrong in Arduino kingdom");

Yardie-, 2019-11-05:
This fixes it but I am not 100% sure of any side effects
 Wire.h add
```
  void onReceive( void (*)(size_t) ); 
```
and Wire.cpp

```
// sets function called on slave write

//~ void TwoWire::onReceive(void(*function)(int)) {
  //~ TinyWireS.onReceive(function);
//~ }

void TwoWire::onReceive( void (*function)(size_t) ) {
  TinyWireS.onReceive(function);
}

void TwoWire::onReceive( void (*function)(int) ) {
  // arduino api compatibility fixer:
  // really hope size parameter will not exceed 2^31 :)
  static_assert(sizeof(int) == sizeof(size_t), "something is wrong in Arduino kingdom");
  TinyWireS.onReceive(reinterpret_cast<void(*)(size_t)>(function));
}

```

Yardie-, 2019-11-23:
Now we are getting fallthrough errors in the twi.c code
I know it is may not be a bug but it is annoying.
It also maybe further upstream from your code.
These can be simply alleviated by adding 
// fall through
just before the next case statement
ie; 
   // fall through
    case TW_ST_DATA_ACK: // byte sent, ack returned

I am not trying to be smart here. I really appreciate your work. 
It simply sent me on the trail of solving this for myself and I as I found the solution I thought I would share it.
I like to leave the warnings as I sometimes forget and also break the rules sometimes, so this is a cool solution.
Thanks for all your brilliant work.

SpenceKonde, 2019-11-25:
The Wire.h and SPI.h libraries included with this core are made specifically for this core. It was made from specific forks of other libraries (for I2C, it's an unholy combination of Wire, TinyWire (USI), TinyWireS (I2C slave), and some software I2C master library for the parts with neither a USI nor a hardware I2C master). These libraries it was based on are generally not being maintained at this point, but even if they were, they're not being monitored for updates to pull into ATTinyCore

So fixes for the Wire and SPI library included with this core should be directed here. Where does that change you mentioned in your last comment need to go? (of course, if you'd like to submit a PR with these fixes, I'd of course welcome that, and then you'd get "credit" for the changes in the commit history and github's "contributor" list)

Yardie-, 2019-11-25:
Ok thanks.
Not sure what a PR is I guess it's a patch request (dinosaur me).

The specific one mentioned was 
line 440 twi.c
```
    // Master Receiver
    // fall through
    case TW_MR_DATA_ACK: // data received, ack sent
```
but it doesn't seem to through it anymore.  I'm not sure what I changed.
There are a couple of other more interesting ones I will track down. If you don' find them first.
 
I like that trick.

SpenceKonde, 2019-11-25:
PR is a "pull request" in github terminology - a "request" that the maintainer of the repo "pull" the changes in the PR into their version of the repository. Created easily through the github web interface (a quick googling will give guides - the exact process will, naturally, depend on whether you're using the github desktop client, command line client, or web interface). Of course, if you have a set of changes, you can always just tell me what they are and have me put them in. 

Either way, I'll be thankful - I'm absolutely swamped right now and don't know when I'll be able to investigate this.

SpenceKonde, 2019-12-11:
Do you happen to have a sample of code which reproduces this problem, so I can check that my fixes work?

Yardie-, 2019-12-12:
Yep the standard examples of Wire slave_receiver and slave_sender
 throw this one on the Attiny85 (no bootloader) and the 841 (no bootloader)(either of the Slave modes) haven't tested others
all the other compile settings are the default values

```
// Wire Slave Receiver
// by Nicholas Zambetti <http://www.zambetti.com>

// Demonstrates use of the Wire library
// Receives data as an I2C/TWI slave device
// Refer to the "Wire Master Writer" example for use with this

// Created 29 March 2006

// This example code is in the public domain.

#include <Wire.h>

void setup() {
  Wire.begin(8);                // join i2c bus with address #8
  Wire.onReceive(receiveEvent); // register event
  Serial.begin(9600);           // start serial for output
}

void loop() {
  delay(100);
}

// function that executes whenever data is received from master
// this function is registered as an event, see setup()
void receiveEvent(int howMany) {
  while (1 < Wire.available()) { // loop through all but the last
    char c = Wire.read(); // receive byte as a character
    Serial.print(c);         // print the character
  }
  int x = Wire.read();    // receive byte as an integer
  Serial.println(x);         // print the integer
}

```

SpenceKonde, 2019-12-12:
hmm, i get the warning, but the code compiles successfully, since the core is using -fpermissive to permit those sort of things - that fix essentially does the same thing that the compiler does with -fpermissive. It changes the function to take a size_t, and when called with an int, casts it to a size_t and calls the version of the function that was changed to take a size_t with that.

SpenceKonde, 2019-12-12:
i guess what i'm wondering is:

Is that stopping the compiling for you, or does it compile with a warning in the console?
Does your change alter the behavior of compiled sketch, or just remove the warning?

Yardie-, 2019-12-12:
Ok if the compiler sorts it out then it's not a problem just annoying. I haven't pushed the system but it does compile and run fine. As far as I can see.

SpenceKonde, 2019-12-12:
Okay, so this now makes sense. I will test your fix and add it to the core
if it doesnt break anything. I just wanted to make sure I was seeing the
same things you were before proceeding (getting rid of warnings is still a
worthy goal; I'd like to get the core to where there are no warnings coming
from the core itself)

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2019-12-14:
okay, I have observed the compile error.

commit 6da30316 (2019-12-14): Fix wire compile issue on x41, 828

fixes #363
closed 2019-12-14 by SpenceKonde