SpenceKonde/ATTinyCore #364: ATtiny167 hardware RXD stop working.

hui5368, 2019-11-12:
Hi,
I'm using Arduino nano as a programmer for ATtiny167.
My project is motor control and communication with UART protocol.
ATtiny167 (pin RXD(PA0) )  will receive the  10 bytes UART packet from the PC and control the motor. 
ATtiny167 (pin TXD(PA1) )  will send 10 bytes UART packet to the PC every 10 seconds.

It's work fine if the ATtiny167 receives 1 time 10 bytes UART packet from the PC every 500ms, but if I try to send 100 times of 10 bytes UART packet to ATtiny167 in 500ms, the ATtiny167 hardware RXD will stop working, but the hardware TXD still works fine, PC side can still receive the UART packet from ATtiny167's TXD.  The hardware RXD will stop working until i restart the ATtiny167.

ATtiny167 using Internal 8Mhz Clock.
Arduino IDE 1.8.10

If i using Arduino Nano in my project everything works fine.

Even i use the following code to check the ATtiny167 hardware RXD, only the hardware RXD will stop working (If I try to send 100 times of 10 bytes UART packet to ATtiny167 in 500ms), but if i use Arduino Nano works fine in this situation.

`const int status_s = 2; // LED
uint8_t rb_rgst;   // read RX buffer

void setup() {
  Serial.begin(9600);
  pinMode(status_s, OUTPUT);
}
void loop() {
  if (Serial.available() > 0) {
    digitalWrite(status_s, HIGH);  // LED will turn on if receive any byte from RX buffer.
      rb_rgst = Serial.read();  //Read the RX buffer 
  }
    digitalWrite(status_s, LOW);  // LED turn off
}`

Does anyone know what is the problem ?

SpenceKonde, 2019-11-15:
A strange one... 

What are you using to generate the serial signal you're feeding it? (asking so that I can be sure I'm reproducing a signal with the same properties as the one you're using).

hui5368, 2019-11-18:
Thanks for your reply :)

I think i found the problem.
In my schematic, I use another MCU to communicate with ATtiny167.
If i sending the data too fast it will make the ATtiny167 RXD stop working (BUT TXD and GPIO still working).

If i remove another MCU, direct sending the data to ATtiny167 through USB/UART Dongle, everything works fine again.

maybe the other MCU corrupts the data. 

Still working on it  :o  :o  :o  :o

SpenceKonde, 2019-11-20:
Interesting! 

IMO, the key thing to determine is whether the issue is that the RX on the 167 breaks, or whether it's the TX on the thing sending the serial data to it that's breaking. You could test this simply as follows:

Connect a serial adapter's GND pin to a GND pin on the 167; ensure that this adapter is connected to your computer.
Start up your test setup, with the other microcontroller sending serial data too fast for physics to permit. Reproduce the bugged state. 
A wire will be going from the RX line of the 167 to the TX line of the other microcontroller. Disconnect this from the TX line of that other micro, and connect it to the TX line of the aforementioned serial adapter. 
Now, open the serial port from an application on the computer, and see how RX is behaving. 

Another quick sanity check would be to connect TX and GND of the other microcontroller to RX and GND of a serial adapter and open port, and see if it's sending correct data or gibberish down the line.