SpenceKonde/ATTinyCore #365: Using a RF 433Mhz with ATtiny45 with RH_ASK.h

josephernest, 2019-11-14:
Hello,
I'm using ATTinyCore for my ATTiny45, and I would like to use a [433Mhz RF transmitter/receiver like this](https://randomnerdtutorials.com/rf-433mhz-transmitter-receiver-module-with-arduino/). 

I've copied https://github.com/PaulStoffregen/RadioHead into `C:\Users\User\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.3.1\libraries`.

When doing:

    #include <RH_ASK.h>
    #include <SPI.h> // Not actually used but needed to compile
    RH_ASK driver;
   
    void setup()
    {
    }

    void loop()
    {
        const char *msg = "Hello World!";
        driver.send((uint8_t *)msg, strlen(msg));
        driver.waitPacketSent();
        delay(1000);
    }

it fails because of the `RH_ASK.h`.

What solution is there to use a RF 433Mhz wih an ATtiny45?

SpenceKonde, 2019-11-15:
I can't offer further advice without seeing a copy of the error output.

SpenceKonde, 2019-12-13:
If you can post the error output, I would be happy to look into this.

SpenceKonde, 2019-12-13:
Alternately, let me know what version of RH_ASK.h you are using (link to where it came from) - there are many forks and versions of the RadioHead library floating around the web, unfortunately.

SpenceKonde, 2019-12-14:
I have reproduced the error. That library (at least that version of it) is not compatible with the attiny parts.

closed 2019-12-14 by SpenceKonde

josephernest, 2019-12-17:
Thank you for your replies, your investigation and sorry for the delay!
IIRC, the error came from Serial indeed, but I don't remember the exact error.

Yes, it seems that it's not made for the ATtiny.

What other RF library would you recommend on ATtiny?

I finally used `Manchester.h`: http://mchr3k.github.io/arduino-libs-manchester/ https://github.com/mchr3k/arduino-libs-manchester and it works with cheap "superheterodyne" modules from ebay.

**What would be your RF recommendations?**