SpenceKonde/ATTinyCore #368: Can't install through boards manager

wvmarle, 2019-11-17:
There's an error in the http://drazzy.com/package_drazzy.com_index.json file.

(firefox:)
SyntaxError: JSON.parse: end of data when ',' or ']' was expected at line 1134 column 1 of the JSON data

(IDE:)
Error downloading http://arduino.esp8266.com/stable/package_esp8266com_index.json http://drazzy.com/package_drazzy.com_index.json
Skipping contributed index file C:\Users\Wouter\AppData\Local\Arduino15\package_drazzy.com_index.json, parsing error occured:
com.fasterxml.jackson.databind.JsonMappingException: Unexpected end-of-input: expected close marker for Array (start marker at [Source: (FileInputStream); line: 2, column: 15])
 at [Source: (FileInputStream); line: 1134, column: 6453]
 at [Source: (FileInputStream); line: 1133, column: 1] (through reference chain: com.fasterxml.jackson.module.mrbean.generated.cc.arduino.contributions.packages.ContributionsIndex["packages"])
Skipping contributed index file C:\Users\Wouter\AppData\Local\Arduino15\package_drazzy.com_index.json, parsing error occured:
com.fasterxml.jackson.databind.JsonMappingException: Unexpected end-of-input: expected close marker for Array (start marker at [Source: (FileInputStream); line: 2, column: 15])
 at [Source: (FileInputStream); line: 1134, column: 6453]
 at [Source: (FileInputStream); line: 1133, column: 1] (through reference chain: com.fasterxml.jackson.module.mrbean.generated.cc.arduino.contributions.packages.ContributionsIndex["packages"])
Skipping contributed index file C:\Users\Wouter\AppData\Local\Arduino15\package_drazzy.com_index.json, parsing error occured:
com.fasterxml.jackson.databind.JsonMappingException: Unexpected end-of-input: expected close marker for Array (start marker at [Source: (FileInputStream); line: 2, column: 15])
 at [Source: (FileInputStream); line: 1134, column: 6453]
 at [Source: (FileInputStream); line: 1133, column: 1] (through reference chain: com.fasterxml.jackson.module.mrbean.generated.cc.arduino.contributions.packages.ContributionsIndex["packages"])
Skipping contributed index file C:\Users\Wouter\AppData\Local\Arduino15\package_drazzy.com_index.json, parsing error occured:
com.fasterxml.jackson.databind.JsonMappingException: Unexpected end-of-input: expected close marker for Array (start marker at [Source: (FileInputStream); line: 2, column: 15])
 at [Source: (FileInputStream); line: 1134, column: 6453]
 at [Source: (FileInputStream); line: 1133, column: 1] (through reference chain: com.fasterxml.jackson.module.mrbean.generated.cc.arduino.contributions.packages.ContributionsIndex["packages"])

alexey1234, 2019-11-17:
I have troubles also

When I install Attiny manually, a have next
`Skipping contributed index file C:\Users\agani\AppData\Local\Arduino15\package_drazzy.com_index.json, parsing error occured:
com.fasterxml.jackson.databind.JsonMappingException: Unexpected end-of-input: expected close marker for Array (start marker at [Source: (FileInputStream); line: 2, column: 15])
 at [Source: (FileInputStream); line: 1134, column: 6453]
 at [Source: (FileInputStream); line: 1133, column: 1] (through reference chain: com.fasterxml.jackson.module.mrbean.generated.cc.arduino.contributions.packages.ContributionsIndex["packages"])
Could not find boards.txt in C:\Users\agani\Documents\Arduino\hardware\Arduino_STM32\examples. Is it pre-1.5?
Could not find boards.txt in C:\Users\agani\Documents\Arduino\hardware\attiny\variants. Is it pre-1.5?`

I updated Arduino few day ago to 1.8.10.  Before update library worked

SpenceKonde, 2019-11-17:
Will fix later today, looks like I messed up {}s in the json file

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2019-11-17:
Fixed now.

closed 2019-11-17 by SpenceKonde

mblanc443, 2024-02-22:
I'm getting following on Arduino 2.3.2
Some indexes could not be updated. Get "https://drazzy.com/package_drazzy.com_index.json": tls: failed to verify certificate: x509: certificate has expired or is not yet valid: : http://drazzy.com/package_drazzy.com_index.json
Please help to resolve this?

sashrikad, 2024-02-26:
Until it's fixed, I hosted the json in github. You can use below url instead:
```
https://raw.githubusercontent.com/sashrikad/drazzy_com/main/index.json
```

Garsti71, 2024-02-26:
Thanks @sashrikad for hosting! That made DxCore possible for me!

JHR1984, 2024-07-14:
Para mi que soy novato tambien ha sido una gran ayuda. Gracias

JHR1984, 2024-07-14:
Buenas 
Yo tengo el mismo problema y estoy intentando programar un attiny 85.
¿Alguna idea de como hacerlo?