SpenceKonde/ATTinyCore #369: Make TX/RX pin on ATTiny85 (and others) configurable

tong67, 2019-11-20:
TinySoftwareSerial.cpp creates Serial object with AIN0 and AIN1.
With this configuration the TX conflicts with I2C. When AIN0 and AIN1 are swapped I2C can be performed and Serial output can be used. This requires the Serial argument to be swapped.
A configuration option to select RX and TX for be AIN0/AIN1 or AIN1/AIN0 would be handly/usefull

SpenceKonde, 2019-11-20:
The builtin software serial actually works with any pin - it's just RX that *only* works on that one specific pin (because it uses the AC interrupt, in order to not use the PCINT interrupt. But I can't take the pin parameter from within the sketch, because Serial.begin() for normal hardware serial, which I'm trying very hard to maintain compatibility with, doesn't take an argument to specify the TX pin (understandable, since this isn't selectable on hardware serial). 

Is there a reason why you can't use the SoftwareSerial library? That library allows you to use any pins (at the cost of using PCINTs for receiving, instead of the AC interrupt, the PCINTs being far more useful than the AC interrupt). Using SoftwareSerial is my recommended solution if you need to create a software serial port on pins other than the ones used by the tinySoftwareSerial "Serial" object.

closed 2019-11-20 by SpenceKonde

E1k3, 2020-03-08:
I noticed, that using the default SoftwareSerial implementation instead of the built-in (Software) Serial requires more memory and more SRAM. I don't know if this is caused by the built-in implementation being smaller than SoftwareSerial or if the built-in is still included, when it is not used.

And besides this, I would like to be able to keep RX on the AC interrupt pin, but move transmit to another one, so that I²C and Serial are possible at the same time. Since SoftwareSerial relies on PCINTs, it won't let me do that (as far as I know).

Regarding compatibility, wouldn't it be possible to add a parameter with a default value (similar to the `config` parameter of the normal hardware serial implementation)? This would not break any existing code that is compatible with the normal implementation.