SpenceKonde/ATTinyCore #370: 841 Alternate MISO MOSI

Yardie-, 2019-11-22:
This is more a feature request or simply a request for more information (or at least a lead to something helpful). 
On reading the 841 data-sheet there seems to be alternative pins for MISO and MOSI.
I am trying to run both I2C and SPI on the same chip and I can't find how to use the alternatives for MOSI and MISO. 
I think that I read somewhere in your documentation that USI SPI would disable USI I2C. 
Is that a pin issue or a USI one. 

Is it possible to run as a SPI master and I2C slave on the same chip. 
Is it possible to use the alternative pin for MOSI so the I2C bus stays separate.
Or is it simply that the the I2C bus will basically stop working whilst I am transmitting over the SPI 
when the MOSI and the SDA are sharing the same pin.

The available information is quite sparse. 
Thanks.

SpenceKonde, 2019-11-22:
On parts with a USI, the USI can only be used for I2C or SPI at a given time, not both - there is only one USI, and it can be configured in two-wire (I2C) or three-wire (SPI) mode, but not both (not only do they share pins, they share the hardware that clocks data in/out. 

However, that fact is irrelevant on the 841, as it does not have a USI - it has hardware I2C slave, and a master/slave SPI interface (the version of the Wire.h library included with this core uses a software I2C implementation on the same pins if you ask for I2C master, and the hardware I2C slave peripheral if you ask for I2C slave - either way, be sure to select the appropriate options from the tools menu for I2C mode - I was never able to keep the master and slave implementations from both hogging flash/ram, even when only one was used). SPI and I2C can both be used at the same time. 

The SPI pins can be moved to the alternate ones by setting the SPIMAP bit in the REMAP register. You may also have to manually set pinMode() on SCK and MOSI (for master) and MISO (for slave), since the SCK/MISO/MOSI #defines aren't updated based on that register. Also double-check which pin is used for SS when the pins are remapped and make sure it is not set as input, as if it's input and goes low (including because it's floating and transitioning randomly) that will switch it into slave mode, as always when using SPI on a classic AVR.

Yardie-, 2019-11-23:
Thank you so much 
that is most of the information I needed. 

At the moment the process of setting the REMAP register eludes me a little.
I see that both are set with 
#define REMAP   _SFR_MEM8(0x65)
and
#define SPIMAP  1
in ./packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/avr/include/avr/iotn841.h
but resetting them appropriately ???

However 
>  SPI and I2C can both be used at the same time.

So if I understand it correctly setting the 841 as I2C slave and implement the SPI as master then they should just work together. without screwing up the I2C bus when the MOSI/SDA pin are using the same pin but I am talking to the SPI slave.

SpenceKonde, 2019-11-23:
REMAP|=(1<<SPIMAP);

Maybe I wasn't clear about SPI and I2C working at the same time - they will
work at the same time, but not if they are using the same pins (the SPI
device will be fine, since it will ignore the data lines as long as CS is
high, but the I2C device will get confused) - so you do need to move the
SPI interface to the alternate pins).

Yardie-, 2019-11-23:
Excellent you are an absolute legend.
This is pure gold and hard to find elsewhere.
Thanks heaps.

closed 2019-11-23 by Yardie-

SpenceKonde, 2019-11-23:
Glad to hear it!

The #defines in io*.h let you treat registers like variables and read/write values to them in that way; some common idioms are:
REGNAME |= (1<<BITNAME); //set a bit
REGNAME &= ~(1<<BITNAME); //clear a bit

Similarly, if you know what you want all bits in a register to be set to, you can just assign that value to it (eg, REGNAME=0x1D; //set REGNAME to 0x1D aka 0b00011101). 

Are you using a crystal as the clock source? If using the internal oscillator, you'll get much less drift if you moved to a crystal.

Yardie-, 2019-11-24:
Thanks Heaps
That is excellent to know I didn't know you could do that. 
Crystal ... Yes I had thought about it maybe but I will have an internal network time server pi which connects to an NTP server. I will push a timestamp regularly. But now I have worked out how to turn TTFonts into rle C bytes the I decode for the ePaper  and I can fit the numbers and a few characters and the code onto an 841. It sounds like a crystal would be a good idea for an independent low power ePaper clock. Maybe some Christmas presents. Haha. 
Anyway thanks heaps. You have opened some new doors for me. 

Greg