SpenceKonde/ATTinyCore #371: Software Serial pins - ATTiny88

finnbear, 2019-11-24:
https://github.com/SpenceKonde/ATTinyCore/blob/46599709e2b69705d37d67b0c88eed1c769bfae7/avr/variants/tinyX8/pins_arduino.h#L56

In this line, and in your extremely useful pinout diagram, you suggest particular TX and RX pins for use with Software Serial. I just finished testing Software Serial (`softwareSerialInstance.println()`) on two other pins, 4 and 5, which worked.

Is there any reason to use the pins you suggest? Or are the pins you suggest used if the code uses non-Software Serial (`Serial.println()`) commands?

Thank you!

SpenceKonde, 2019-11-24:
(Note: "SoftwareSerial" refers to the included library of that name, which
let's you create a software serial port using any pins. "software serial"
(with space between the words and no caps) refers to any implementation of
serial port functionality that doesnt require a hardware serial port.)

The tiny88 does not have hardware serial. 'Serial' on parts without a
hardware serial port is a software serial implementation on those pins.
Those pins are used so that it can use the analog comparator interrupt to
react to transitions on the incoming serial line. In contrast, the
SoftwareSerial library provides a software serial implementation that will
work on any pin, but it uses PCINTs - in order to work on all pins, it
takes over all PCINTs (even though only one bank of them, the one
containing the RX pin, is needed, the compiler isnt smart enough to realize
this and not define the interrupt handlers for all banks), meaning that you
cant use PCINTs for other purposes nor can it use libraries that use them.

So, your choices if you want a serial port on the t88 (or other supported
tiny without a hardware serial port) are to use Serial on those specific
pins, which uses the analog comparator interrupt (this is generally not a
very useful interrupt, so unlikely to conflict with other code/libraries),
or use SoftwareSerial which works on any pins you choose, at the cost of
using up the extremely useful PCINT interrupts.

Either way, you have a software serial port (either named Serial, or
whatever you named the instance of SoftwareSerial). The choice is just
between using those specific pins for serial (and being able to use PCINTs
elsewhere in the sketch), or using whichever pins you want for serial (but
not being able to use PCINTs for anything else).

Note that software serial ports, regardless of how they are implemented,
are half duplex (cannot transmit while receiving or vice-versa), sending is
blocking, and baud rates should be kept on the low side.

If using SoftwareSerial library, there is nothing special about those pins
- they are only relevant when using the built in software serial port
'Serial'.

Hope this helps
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

finnbear, 2019-11-24:
Yeah, this helps. This helps quite a lot. Thank you!

closed 2019-11-24 by finnbear