SpenceKonde/ATTinyCore #373: Documentation suggestion - Burn Bootloader

technoblogy, 2019-12-03:
You have the following important information in your Quick Gotcha list section:

> When using a chip for the first time, or after changing the clock speed or BOD settings, you must do "burn bootloader" to set the fuses, even if you are not using the chip with a bootloader

Can I suggest that you repeat this in the section "Changing the ATtiny clock speed and other settings" where you say:

> Changing the ATtiny clock speed, B.O.D. settings etc. is made easy:

This section makes it sound like you can change the internal clock speed just by choosing a different item from this menu.

finnbear, 2019-12-04:
To add to this issue, I seem to remember you saying that the option to use an external clock is not exposed to avoid bricking the ATTiny if one is not present.

> DANGER if this clock source is not present, you must supply a clock source to CLKI pin before it can be reprogrammed, including to use a different clock source. Unlike the parts which support a crystal, where this "potentially dangerous" clock option is not available in the menus because there are menu options for the supported speeds with a crystal as a clock source, so the procedure below can be used.

This [post](https://forum.arduino.cc/index.php?topic=384686.msg2651885#msg2651885) says otherwise: 
> having designed a few programmers i find a crystal for the target is not needed

So my question is: If I don't include any external crystal/clock on my ATTiny/ATMega design, will it still be programmable over ICSP (with SCK) even if its fuses were set to use an external clock? Does
> CLKI

refer to XTAL1, XTAL2, or SCK?

technoblogy, 2019-12-04:
If you have set the fuses for an external crystal you won't be able to program with ISP if a crystal isn't present, even to change the fuses back to an internal clock. By the way, I find that putting a crystal across the XTAL1 and XTAL2 pins, with no capacitors, usually works OK to get out of this.

If you had set the fuses for an external clock, you would need to provide a clock signal on CLKI to program with ISP, so it is potentially dangerous. CLKI is the same pin as XTAL1.

SpenceKonde, 2019-12-04:
Thanks for bringing this up - this is the kind of thing I love to have reported in the docs, because I am deep enough down the avr rabbit hole that I don't always realize what will be confusing for someone who is relatively new to these parts.

technoblogy, 2019-12-04:
You're welcome!

SpenceKonde, 2019-12-13:
Documentation updated to include mention of this in the place you pointed out.

closed 2019-12-13 by SpenceKonde