SpenceKonde/ATTinyCore #374: Compilation error

classic-audio, 2019-12-04:
I have installed the ATTinyCore, I get an error when I try to compile my sketch.
This is the error message
`
Arduino: 1.8.10 (Windows 10), Board: "ATtiny25/45/85, Enabled, CPU, ATtiny85, 8 MHz (internal), EEPROM retained, B.O.D. Disabled, Enabled"

In file included from C:\Users\Frede\Dropbox\ALL  -  P R O J  E C T S\Grammofon-IR-Tacko\Project-2019-11-V2.0-RPM-OLED-ATtiny85-TinyOzOLED\Project-2019-11-V2.0-RPM-OLED-ATtiny85-TinyOzOLED.ino:13:0:

C:\Users\Frede\Documents\Arduino\libraries\Tiny4kOLED/Tiny4kOLED.h:15:1: error: 'https' does not name a type

 https://github.com/adafruit/TinyWireM

 ^~~~~

Multiple libraries were found for "Tiny4kOLED.h"
 Used: C:\Users\Frede\Documents\Arduino\libraries\Tiny4kOLED
Multiple libraries were found for "ATTinyCore.h"
 Used: C:\Users\Frede\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.3.2\libraries\ATTinyCore
exit status 1
Error compiling for board ATtiny25/45/85.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

What is my problem?
I hope for some help to to this to work.

SpenceKonde, 2019-12-04:
Looks like a problem in that library file, my guess would be that that url
is supposed to be a comment but isn't.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2019-12-04:
Either that, or that library is intending to use a different library also named ATTinyCore, but the IDE is picking the ATTinyCore library that comes with this core? 
Unfortunately a bug in the 1.8.10 IDE mangles the list of libraries found in the console output.

SpenceKonde, 2019-12-04:
But yeah, I think most likely this is a syntax error in the library, but I can't confirm that without knowing where the library came from (name does not uniquely identify libraries due to them almost invariably being hosted in github, and hence forkable by anyone)

SpenceKonde, 2019-12-04:
Let me know regarding the library - closing this because at present it does not appear to be a bug in the core, but rather in the library (I will still receive notifications, and will reopen if further information indicates a problem in the core)

closed 2019-12-04 by SpenceKonde