SpenceKonde/ATTinyCore #375: Documentation Suggestion - Wiring - Accidentally pulling down SCK/MOSI/MISO

finnbear, 2019-12-05:
On one of my designs, I narrowed down a problem preventing ICSP programming to the fact that I had attached an led from the SCK pin of my ATTiny(85) to ground through another resistor. Apparently SPI is active-HIGH and this LED/Resistor prevents SPI from working with my USBTinyISP programmer (by pulling SCK low).

My suggestion: Add "Make sure MISO, MOSI, and SCK are not connected to GND, even though a resistor or LED" to your wiring guide.

Feel free to dismiss this issue if you are sure that I misdiagnosed the problem.

SpenceKonde, 2019-12-05:
Thanks, yeah, will add something like that. Note that how much of a load on
the sck and mosi pins ir takes to interfere depends on the design of the
programmer. Most have series resistors (to protect programmer from shorts,
and allow a 5v programmer to program an (externally powered) target running
at a lower voltage) on the data lines, but the value of these resistors
varies between programmers.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2019-12-05:
Reopening until I actually make the changes, lest this get lost

SpenceKonde, 2019-12-13:
A new Programming Guide has been added which covers this. See Programming.md

Thanks for reporting this - it led to what I think is a much better guide to programming these parts.

closed 2019-12-13 by SpenceKonde