SpenceKonde/ATTinyCore #376: ATTiny 88 cannot burn bootloader

finnbear, 2019-12-06:
I am trying to use my USBTinyISP (SparkFun) programmer to program an ATTiny 88 board that I designed. Programming the sketch directly (no bootloader) works. Burning the "bootloader" with the bootloader disabled (so only fuses) works. When I try to use the optiboot mode, I get this:
```
Arduino: 1.8.10 (Linux), Board: "ATtiny48/88 (optiboot), Enabled, ATtiny88, 8 MHz (internal), EEPROM retained, B.O.D. Disabled, Enabled, Port A (pins 23~26)"

free(): invalid next size (normal)
Error while burning bootloader.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```
I have tried on two separate computers and I get the same error.

Any ideas?

SpenceKonde, 2019-12-06:
Thanks for the report, I will attempt to reproduce and see if I can figure out what went wrong. 

Which version of the core is this with? The current github version may have problems, iirc I was halfway through making some significant improvements to this area when the thanksgiving holiday and some commissioned designs for clients struck and I havn't had time to finish it.

finnbear, 2019-12-06:
Alright. My version is the latest one from the IDE board manager, v1.3.2

SpenceKonde, 2019-12-06:
Well that doesn't sound right! Will see if i can reproduce, hopefully tonight.

SpenceKonde, 2019-12-07:
Hmmmm

I was unable to reproduce this issue on Windows 10/1.8.9 and USBTinyISP (ATTinyCore) as the programmer. 

However, I don't have ready access to a linux system to try it on...

SpenceKonde, 2019-12-07:
Did you select USBTinyISP (ATTinyCore) as the programmer (not USBTinyISP - due to the weird behavior of the Arduino IDE, if you select a programmer defined by another core, it will ignore the upload recipe provided by the third party core and use the one for the default core, which uses a version of avrdude.conf that has issues with some attiny parts. 

In preferences, under "Show verbose output during", check "upload" (not compilation - the error message the IDE shows is wrong, a fact which was reported during the summer and has been ignored by the Arduino team ever since) and post the output you get here?

finnbear, 2019-12-07:
I can confirm I selected USBTinyISP (ATTinyCore) as the bootloader
```
Arduino: 1.8.10 (Linux), Board: "ATtiny48/88 (optiboot), Enabled, ATtiny88, 8 MHz (internal), EEPROM retained, B.O.D. Disabled, Enabled, Port A (pins 23~26)"

/home/finnb/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/home/finnb/.arduino15/packages/ATTinyCore/hardware/avr/1.3.2/avrdude.conf -v -pattiny88 -cusbtiny -e -Uefuse:w:0xfe:m -Uhfuse:w:0b11010111:m -Ulfuse:w:0xE2:m 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/finnb/.arduino15/packages/ATTinyCore/hardware/avr/1.3.2/avrdude.conf"
         User configuration file is "/home/finnb/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbtiny
         Setting bit clk period        : 5.0
avrdude: usbdev_open(): Found USBtinyISP, bus:device: 001:035
         AVR Part                      : ATtiny88
         Chip Erase delay              : 15000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no         64    4      0  3600  3600 0xff 0xff
           flash         65     6    64    0 yes      8192   64    128  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude: programmer operation not supported

avrdude: Setting SCK period to 5 usec
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9311 (probably t88)
avrdude: erasing chip
avrdude: Setting SCK period to 5 usec
avrdude: reading input file "0xfe"
avrdude: writing efuse (1 bytes):

/home/finnb/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/home/finnb/.arduino15/packages/ATTinyCore/hardware/avr/1.3.2/avrdude.conf -v -pattiny88 -cusbtiny -Uflash:w:/home/finnb/.arduino15/packages/ATTinyCore/hardware/avr/1.3.2/bootloaders/optiboot/optiboot_attiny88_8000000L.hex:i 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/finnb/.arduino15/packages/ATTinyCore/hardware/avr/1.3.2/avrdude.conf"
         User configuration file is "/home/finnb/.avrduderc"
[... 94 lines cut ...]
avrdude: writing flash (8192 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 8192 bytes of flash written
avrdude: verifying flash memory against /home/finnb/.arduino15/packages/ATTinyCore/hardware/avr/1.3.2/bootloaders/optiboot/optiboot_attiny88_8000000L.hex:
avrdude: load data flash data from input file /home/finnb/.arduino15/packages/ATTinyCore/hardware/avr/1.3.2/bootloaders/optiboot/optiboot_attiny88_8000000L.hex:
avrdude: input file /home/finnb/.arduino15/packages/ATTinyCore/hardware/avr/1.3.2/bootloaders/optiboot/optiboot_attiny88_8000000L.hex contains 8192 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 8192 bytes of flash verified

avrdude done.  Thank you.

free(): invalid next size (normal)
Error while burning bootloader.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```
Looks like it got all the way to the end but produced an error anyway. I will try checking if the board has a bootloader now...

finnbear, 2019-12-07:
Okay the board definitely has a working boot-loader. I can upload code without incident, but the "error" while burning bootloader is still an issue.

SpenceKonde, 2019-12-07:
I would say that this error is spurious, as it is shown despite successful upload. Weird behavior; it sounds like a bug in avrdude for linux.

finnbear, 2019-12-07:
It doesn't seem to affect other hardware though... :confused:

SpenceKonde, 2019-12-07:
Yeah - barring more information about what this error means - I have no idea how to interpret it - or association with a lack of functionality, I think this is an avrdude issue, not an ATTinyCore issue. I don't think the code with the problem is under my control.

finnbear, 2019-12-07:
Okay, I guess you can close this issue. Thanks for suggesting verbose logging :)

SpenceKonde, 2019-12-07:
Yeah, I'm not really happy with the error output here. I'm going to put it into Documentation so I can be sure to add a warning about it.

sleemanj, 2019-12-07:
https://savannah.nongnu.org/bugs/?48776

SpenceKonde, 2019-12-07:
thanks James!

closed 2019-12-07 by SpenceKonde