SpenceKonde/ATTinyCore #377: Attiny x313 with USB-TTL programmer

sideswiper22, 2019-12-10:
Hello, I'm attempting to design a prototyping board using the ATTiny 2313A chip and uploading using the ATTinyCore. I had a bunch laying around in my workshop and I can buy more for cheap (about 70 cents apiece) Initially, I'm just using a cheap CH340G module to test out if programming through UART works since I cannot find any resources about it.

I connected the Vcc, Rx, Tx, and GND pins to the right pins however, I keep getting errors. I first tried it with AVRISP MkII (ATTinyCore) as the programmer but it would just send these:

Using Port                    : usb
Using Programmer              : stk500v2
Setting bit clk period        : 5.0
An error occurred while uploading the sketch
avrdude: usbdev_open(): did not find any USB device "usb" (0x03eb:0x2104)

Then, I tried AVR ISP (ATTinyCore):

Using Port                    : COM4
Using Programmer              : stk500v1
Setting bit clk period        : 5.0
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x03
^this will go on until attempt 10

At first I thought it was the programmer, so I tested it on my Arduino uno, manually pressing the reset button since the programmer didn't have a DTR breakout from the chip, and it worked fine.

I'm using the ATTiny 2313A with internal resistors. Maybe that's the problem? Tuning the internal oscillator is out of the question since the chip has no ADC's. However, it would take some time to determine this since I'm all out of oscillators and I have to buy new ones.

Or maybe I was wrong in my understanding that the ATTiny 2313 could work with this method of programming?

SpenceKonde, 2019-12-10:
Programming via uart requires a bootloader (optiboot). ATTinyCore does not
include optiboot for parts with less than 8k of flash - because it has to
use virtual boot, it eats like 640b of flash; on a 2313, that doesnt leave
much space for your code! For those parts, you need to use an ISP
programmer (also need that to install bootloader and set clock source on
any classic avr)

If you want 70-cent AVRs, check out the new attiny1614 (the 3216 and 1606
have more pins and arent much more expensive, and there are smaller-flash
versions for less). They have the super capable new megaavr peripherals,
built in 16/20MHz oscillator, uart, i2c, spi, more pwm pins... 1614 is 71
cents a pop from digikey with 11 usable I/O pins, 16k flash, 2k sram, 6 pwm
pins.

That whole family of new megaavr attiny parts is supported on the arduino
IDE by my other core, megaTinyCore. You need to use an arduino nano or
something as a UPDI programmer (instructions in the readme) but my core
includes optiboot for them, and it has a 512b serial bootloader
(optiboot-x), so if using that youd only need to use it once per chip to
bootload it and could then upload via serial. The bootloader supposedly
even does eeprom writes, unlike the 512b optiboot bootloader for classic
avrs!

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2019-12-11:
Closing because not an issue in attinycore. Will still receive notifications, feel free to ask further questions.

closed 2019-12-11 by SpenceKonde

sideswiper22, 2019-12-11:
Well, it looks like my current project is on a dead end, but thank you so much for your detailed response and advice! I can hopefully find another use for my 2313's. I have also taken the time to look into the other ATTiny's you've mentioned and I'm on my way to alter the design to accommodate them. They do seem to be better AVRs overall for the price, especially since some of them have built-in DACs!

I apologize for the late response.

sideswiper22, 2019-12-13:
Hello! I've been reading a bit on the megaTinyCore and I'm confused about the configuration of the autoreset circuit. Is the signal diode in series with the resistor, or parallel to it, connecting directly to Vcc? Also, what is the purpose of this diode? Reset circuity doesn't usually have this extra part.

SpenceKonde, 2019-12-13:
Yes, parallel - with the band towards Vcc. 

When DTR is brought low (when port is opened), because it's capacitively coupled to RST, it will briefly pull reset low (resetting the chip), and then the pullup will pull RST back up to Vcc. Now RST is at vcc. When DTR is released, DTR will jump back up to Vcc. It is capacitively coupled to RST, so RST will briefly jump up to 2x Vcc. The RST pin is 12v tolerant to permit HV programming - but that's also the problem. On the megaavr parts*, this would trigger the 12v UPDI pin function override, turning the pin from reset back into UPDI until next power cycle, which would obviously interfere with functionality. This diode prevents that from happening. 

That diode is also recommended whenever the DTR reset circuit is in use (it is present on official boards), even on classic AVRs, due to a concern about briefly entering HV programming mode. 

Note that this circuit will prevent UPDI programming from working (due to the resistor), and in order for autoreset to work, you must "burn bootloader" with it set to use UPDI as reset, which disables further programming via UPDI. So some care is needed with selection of options before you go for it. On my boards, the Rev. A boards have a jumper on the back that is initially not connected, so you can install all parts, then burn the bootloader on the assembled board, then connect the reset circuit with a solder bridge across the little jumper on the back.

*Some of them at least, in some silicon revs - apparently some parts in the wild also require a KEY command to be sent over UPDI pin after the 12v pulse. And some don't.

sideswiper22, 2020-01-06:
Regarding the options when installing the serial bootloader, more specifically the clock speeds, are there any drawbacks in choosing 20Mhz over 16Mhz? Is 20Mhz stable enough, and do the libraries fully support 20Mhz speeds without issues?

jLynx, 2022-05-31:
I have been getting ``avrdude: Device signature = 0x000000`` when trying to program my ATTiny 2313A using Arduino as ISP with the following connections:
```
Arduino--------Attiny2313
  |                   |
13------------>19
12------------>18
11------------>17
10------------>1
5v------------>20 VCC
GND-------->10 GND
```
(And a 10uF on GND and reset of the Arduino)
[image]

My question, is am I doing something wrong, or have I been sent fake chips? I have testing using 2 different 2313A's