SpenceKonde/ATTinyCore #379: not recognized as an internal or external command error on ATTiny

jrhilton, 2020-01-03:
Hi, while trying to compile any code for a board using this library I get the error below using debug.

I've tried a clean install of Arduino IDE (including removing all the temp/local data folders etc) but on reinstall it has the same problem. Reinstall consists of downloading the IDE from the website, adding the library link and using boards manager to install ATTinyCore. The out the box boards in the IDE work fine as do adding others like ESP32 etc.

At the start it says "Using board 'attinyx5' from platform in folder: C:\Users\James
Using core 'tiny' from platform in folder: C:\Users\James" which is very odd as that user name is no longer used even though it is still in C:\Users. Our user is James & Milly. I suspect this is the issue but why is it doing this just for this board library when installed through board manager, and how can I fix it?

`C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\James & Milly\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\James & Milly\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\James & Milly\Documents\Arduino\libraries -fqbn=ATTinyCore:avr:attinyx5:LTO=enable,TimerClockSource=default,chip=45,clock=8internal,eesave=aenable,bod=disable,millis=enabled -ide-version=10810 -build-path C:\Users\JAMES&~1\AppData\Local\Temp\arduino_build_325406 -warnings=none -build-cache C:\Users\JAMES&~1\AppData\Local\Temp\arduino_cache_502314 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\James & Milly\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\James & Milly\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=C:\Users\James & Milly\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\James & Milly\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -verbose C:\Users\James & Milly\Documents\Arduino\att45_tst\att45_tst.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\James & Milly\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\James & Milly\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\James & Milly\Documents\Arduino\libraries -fqbn=ATTinyCore:avr:attinyx5:LTO=enable,TimerClockSource=default,chip=45,clock=8internal,eesave=aenable,bod=disable,millis=enabled -ide-version=10810 -build-path C:\Users\JAMES&~1\AppData\Local\Temp\arduino_build_325406 -warnings=none -build-cache C:\Users\JAMES&~1\AppData\Local\Temp\arduino_cache_502314 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\James & Milly\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\James & Milly\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=C:\Users\James & Milly\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\James & Milly\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -verbose C:\Users\James & Milly\Documents\Arduino\att45_tst\att45_tst.ino
**Using board 'attinyx5' from platform in folder: C:\Users\James
Using core 'tiny' from platform in folder: C:\Users\James**
Detecting libraries used...
"C:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=attiny45 -DF_CPU=8000000L -DCLOCK_SOURCE=0 -DARDUINO=10810 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR -DNEOPIXELPORT=PORTB "-IC:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\ATTinyCore\\hardware\\avr\\1.3.3\\cores\\tiny" "-IC:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\ATTinyCore\\hardware\\avr\\1.3.3\\variants\\tinyX5" "C:\\Users\\JAMES&~1\\AppData\\Local\\Temp\\arduino_build_325406\\sketch\\att45_tst.ino.cpp" -o nul
Generating function prototypes...
"C:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=attiny45 -DF_CPU=8000000L -DCLOCK_SOURCE=0 -DARDUINO=10810 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR -DNEOPIXELPORT=PORTB "-IC:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\ATTinyCore\\hardware\\avr\\1.3.3\\cores\\tiny" "-IC:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\ATTinyCore\\hardware\\avr\\1.3.3\\variants\\tinyX5" "C:\\Users\\JAMES&~1\\AppData\\Local\\Temp\\arduino_build_325406\\sketch\\att45_tst.ino.cpp" -o "C:\\Users\\JAMES&~1\\AppData\\Local\\Temp\\arduino_build_325406\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Program Files (x86)\\Arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\JAMES&~1\\AppData\\Local\\Temp\\arduino_build_325406\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=attiny45 -DF_CPU=8000000L -DCLOCK_SOURCE=0 -DARDUINO=10810 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR -DNEOPIXELPORT=PORTB "-IC:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\ATTinyCore\\hardware\\avr\\1.3.3\\cores\\tiny" "-IC:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\ATTinyCore\\hardware\\avr\\1.3.3\\variants\\tinyX5" "C:\\Users\\JAMES&~1\\AppData\\Local\\Temp\\arduino_build_325406\\sketch\\att45_tst.ino.cpp" -o "C:\\Users\\JAMES&~1\\AppData\\Local\\Temp\\arduino_build_325406\\sketch\\att45_tst.ino.cpp.o"
Compiling libraries...
Compiling core...
"C:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=attiny45 -DF_CPU=8000000L -DCLOCK_SOURCE=0 -DARDUINO=10810 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR -DNEOPIXELPORT=PORTB "-IC:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\ATTinyCore\\hardware\\avr\\1.3.3\\cores\\tiny" "-IC:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\ATTinyCore\\hardware\\avr\\1.3.3\\variants\\tinyX5" "C:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\ATTinyCore\\hardware\\avr\\1.3.3\\cores\\tiny\\WInterrupts.c" -o "C:\\Users\\JAMES&~1\\AppData\\Local\\Temp\\arduino_build_325406\\core\\WInterrupts.c.o"
"C:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=attiny45 -DF_CPU=8000000L -DCLOCK_SOURCE=0 -DARDUINO=10810 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR -DNEOPIXELPORT=PORTB "-IC:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\ATTinyCore\\hardware\\avr\\1.3.3\\cores\\tiny" "-IC:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\ATTinyCore\\hardware\\avr\\1.3.3\\variants\\tinyX5" "C:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\ATTinyCore\\hardware\\avr\\1.3.3\\cores\\tiny\\wiring_digital.c" -o "C:\\Users\\JAMES&~1\\AppData\\Local\\Temp\\arduino_build_325406\\core\\wiring_digital.c.o"
"C:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-l [... cut ...]
cmd /C "C:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-objdump" --disassemble --source --line-numbers --demangle --section=.text "C:\\Users\\JAMES&~1\\AppData\\Local\\Temp\\arduino_build_325406/att45_tst.ino.elf" > "C:\\Users\\JAMES&~1\\AppData\\Local\\Temp\\arduino_build_325406/att45_tst.ino.lst"
'C:\Users\James' is not recognized as an internal or external command,

operable program or batch file.

The system cannot find the path specified.

Access is denied.

The system cannot find the path specified.

exit status 1
Error compiling for board ATtiny25/45/85.`

jrhilton, 2020-01-03:
To confirm i'm using IDE v1.8.10 and ATTinyCore v 1.3.3.

SpenceKonde, 2020-01-03:
It looks like that was from export compiled binary - the call to the batch
file that generates the boo tripping over the & in the username. I'm not
sure if theres another value that I can substitute in that will give the
correctly escaped version of that path; I can give no eta on a fix. Normal
compile works, right?

A username that doesn't include special characters will definitely work.

Also, does it happen on 1.8.9? There were some changes in how 1.8.10 works
with file paths (this caused the xp compatibility issue and I think is
getting reverted for that reason.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

jrhilton, 2020-01-04:
Hi, in terms of what I was doing it was a new empty sketch and I just hit Verify.

I've also tried with IDE 1.8.9 with 1.3.3 and also library version 1.3.1 and the same error still comes up. Log below for IDE 1.8.9 and 1.3.1.

This time it has the correct platform folder though -  "Using board 'attinyx5' from platform in folder: C:\Users\James & Milly\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.3.1
Using core 'tiny' from platform in folder: C:\Users\James & Milly\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.3.1"

`Arduino: 1.8.9 (Windows 10), Board: "ATtiny25/45/85, Enabled, CPU, ATtiny85, 8 MHz (internal), EEPROM retained, B.O.D. Disabled, Enabled"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\James & Milly\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\James & Milly\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\James & Milly\Documents\Arduino\libraries -fqbn=ATTinyCore:avr:attinyx5:LTO=enable,TimerClockSource=default,chip=85,clock=8internal,eesave=aenable,bod=disable,millis=enabled -ide-version=10809 -build-path C:\Users\JAMES&~1\AppData\Local\Temp\arduino_build_776014 -warnings=none -build-cache C:\Users\JAMES&~1\AppData\Local\Temp\arduino_cache_454316 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\James & Milly\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\James & Milly\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=C:\Users\James & Milly\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\James & Milly\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -verbose C:\Users\JAMES&~1\AppData\Local\Temp\untitled2067978025.tmp\sketch_jan04b\sketch_jan04b.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\James & Milly\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\James & Milly\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\James & Milly\Documents\Arduino\libraries -fqbn=ATTinyCore:avr:attinyx5:LTO=enable,TimerClockSource=default,chip=85,clock=8internal,eesave=aenable,bod=disable,millis=enabled -ide-version=10809 -build-path C:\Users\JAMES&~1\AppData\Local\Temp\arduino_build_776014 -warnings=none -build-cache C:\Users\JAMES&~1\AppData\Local\Temp\arduino_cache_454316 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\James & Milly\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\James & Milly\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=C:\Users\James & Milly\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\James & Milly\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -verbose C:\Users\JAMES&~1\AppData\Local\Temp\untitled2067978025.tmp\sketch_jan04b\sketch_jan04b.ino
Using board 'attinyx5' from platform in folder: C:\Users\James & Milly\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.3.1
Using core 'tiny' from platform in folder: C:\Users\James & Milly\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.3.1
Detecting libraries used...
"C:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10809 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR -DNEOPIXELPORT=PORTB "-IC:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\ATTinyCore\\hardware\\avr\\1.3.1\\cores\\tiny" "-IC:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\ATTinyCore\\hardware\\avr\\1.3.1\\variants\\tinyX5" "C:\\Users\\JAMES&~1\\AppData\\Local\\Temp\\arduino_build_776014\\sketch\\sketch_jan04b.ino.cpp" -o nul
Generating function prototypes...
"C:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10809 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR -DNEOPIXELPORT=PORTB "-IC:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\ATTinyCore\\hardware\\avr\\1.3.1\\cores\\tiny" "-IC:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\ATTinyCore\\hardware\\avr\\1.3.1\\variants\\tinyX5" "C:\\Users\\JAMES&~1\\AppData\\Local\\Temp\\arduino_build_776014\\sketch\\sketch_jan04b.ino.cpp" -o "C:\\Users\\JAMES&~1\\AppData\\Local\\Temp\\arduino_build_77 [... cut ...]
cmd /C "C:\\Users\\James & Milly\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-objdump" --disassemble --source --line-numbers --demangle --section=.text "C:\\Users\\JAMES&~1\\AppData\\Local\\Temp\\arduino_build_776014/sketch_jan04b.ino.elf" > "C:\\Users\\JAMES&~1\\AppData\\Local\\Temp\\arduino_build_776014/sketch_jan04b.ino.lst"
'C:\Users\James' is not recognized as an internal or external command,

operable program or batch file.

The system cannot find the path specified.

Access is denied.

The system cannot find the path specified.

exit status 1
Error compiling for board ATtiny25/45/85.`

SpenceKonde, 2020-01-10:
I still don't understand why it's calling avr-objdump when it's not trying to do an export compiled binary...

SpenceKonde, 2020-02-14:
Will see if I can reproduce anything like this over the weekend, but I am not optimistic at this point - however, I do have one insight into the error message: there's a bug with some recent IDE versions that is hiding part of the error message and was leading me astray in parts of my analysis above - I no longer think the issue is related to the username, specifically, paths output to the debug window are incorrectly truncated if they contain a space, which makes it look like that's the cause of the problem, when in fact only the output is wrong and the correct path is being used internally. I don't know if the error output issue is fixed in 1.8.11, but it was introduced in a recent 1.8.x version. The key difference is that on versions of the IDE that don't have this bug, instead of 

> 'C:\Users\James' is not recognized as an internal or external command, operable program or batch file.

it will show the entire path, likely:
> 'C:\Users\James & Milly\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5/bin/avr-objdump' is not recognized as an internal or external command, operable program or batch file.

SpenceKonde, 2020-03-02:
Update on investigation - I have confirmed that (to my surprise) those scripts are getting called even when not doing export compiled binary. 

Let me know if you're able to get the full error message with latest version of the IDE

SpenceKonde, 2020-03-02:
FFS, why does github close any issue I even mention in a commit message, except when I want it to. Like I put something like "fixes #blahblahblah" - issue stays open. But if I refer to an issue in passing, even to say it's not related even if it looks like it is - the issue gets auto-closed.

Floessie, 2020-03-03:
Maybe [this](https://help.github.com/en/github/managing-your-work-on-github/linking-a-pull-request-to-an-issue#linking-a-pull-request-to-an-issue-using-a-keyword) sheds some light. Handy feature, and it's obvious, why it triggered...

HTH,
Flössie

SpenceKonde, 2020-03-22:
As requested debugging data has not been provided, I am unable to provide further assistance here. If the full error output can be obtained (using the latest version of the IDE), that may shed new light on this issue. The issue does not reproduce for me, but I am also not willing to go through the effort of setting up a user account on a windows 10 computer with an ampersand ( & ) in it to see if that might be the cause of the issue.

commit 9d2d4ecb (2020-03-02): Will need to check that this hasn't broken export or normal builds...

This would fix #379 if it works (at least insofar as normal builds are concerned - export compiled binary would still be busted, but at least compile would work, and there's no need to run objdump if not generating the listing for export compiled binary.
closed 2020-03-22 by SpenceKonde