SpenceKonde/ATTinyCore #380: Issue with TinySoftSerial

SpenceKonde, 2020-01-06:
Via email:

Hey,
I'm using your TinyCore with Attiny85. It has simply changed the landscape for me. 

It uses TinySoftwareSerial for serial comm. I have observed that the TinySoftwareSerial Lib is not properly implemented. In inherits Stream but doesn't override the member functions properly.

1. the declarations for available, read, peek, and flush should be:
fooReturnType foo(foo args) override;

instead of
virtual fooReturnType foo(foo args);
Although it compiles without errors (I don't exactly know what would be the consequences of that, may be the Stream class doesn't see the new declarations), but still, standards have got to have some value.

2. Secondly, available() isn't working, peek() isn't implemented at all and read() seems to be ignoring the implementation of peek completely.

It should be something like this:

int TinySoftwareSerial::peek(void)
{
      // peekByte is defined as private peekByte = -1 in the class definition.
      if (peekByte < 0)
           peekByte = read();
      return peekByte;
}

int TinySoftwareSerial::read(void)
{
      if (peekByte >= 0)
      {
            int bkp = peekByte;
            peekByte = -1;
            return bkp;
      }
      // ... the rest remains the same.
}

3. I'm currently using peek() instead of available() because I'm still looking into available() for a fix; it always return 0. I'll let you know when it is fixed

sleemanj, 2020-01-06:
> In inherits Stream but doesn't override the member functions properly.

Yes it does.

`override` is just an advisory to a C++11 compiler so that if the author made an error it would fail if the method did not **actually** override anything in other words the keyword `override` means "my intention is that this method is overriding something in a parent class, please Mr Compiler if that's not actually going to happen because I goofed then tell me about it"

The existing method declaration (`virtual fooReturnType foo(foo args);`) is equivalent (and correct for pre C++11).  

Technically `virtual` is not needed in the derived class as it is implied from the parent classe's virtual method being override but it does no harm and is clearer that this method can be overridden if anybody inherits from`TinySoftwareSerial`

With that said I don't know what C++ version the Arduino compilation process targets off the top of my head, I would assume it is at least C++11 and in which case adding the "override" keyword would do no harm and generate a compiler error if the signatures of the underlying Stream virtual methods ever got changed (which seems unlikely, but...).

SpenceKonde, 2020-01-10:
Are you sure you are using my core, not some other one?

Mine definitely has an implementation of peek() and available() - see https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/cores/tiny/TinySoftwareSerial.cpp and the class is named TinySoftwareSerial - if there's a class named TinySoftSerial, that is not the builtin software serial included with my core, nor a library supplied as a library with my core.

umartechboy, 2020-01-27:
I don't think I can confirm this now. I wrote over the original files while making my changes.

I'l dig a deeper into my backups and see what i find. It will take a couple of days due to my current engagements.

SpenceKonde, 2020-02-07:
Closing this as last action was 11 days ago and no updates, assuming this issue was due to different core having been used; if that is not the case, please reopen, and include code which demonstrates the issue.

closed 2020-02-07 by SpenceKonde