SpenceKonde/ATTinyCore #381: Serial communication stopped working with 1.3.3

Sergior01, 2020-01-09:
Hi,

I have a code for the ATTiny841 that uses the serial function. When I upload the code using the 1.3.2 version everything works fine but if I use the 1.3.3 the ATTiny841 can still receive data but it only sends symbols. The UART connects to a RS485 Maxim IC so I also use PA3 to control the transmission/reception of data so I still have to make sure this is not the source of the problem.

I've seen on the changelog that the HardwareSerial function was updated so I suspect it might have introduced some bug. Is anyone else experiencing similar issues?

Thank you,
Sergio

SpenceKonde, 2020-01-10:
Will investigate - the entirety of hardware serial was replaced with a new version to fix a number of outstanding problems and inefficiencies. But the peripherals are the same on the atmega devices used on official boards and the attiny841, so I'm unsure why the behavior would be busted on the tiny...

Have you made sure you are not releasing the tx/rx select pin before transmission is finished? Recall that serial transfers are buffered and each byte is then sent in the background (assuming all bytes will fit in the buffer). If you don't pay attention to this, you'll have problems trying to control an RS485 transceiver with it.

SpenceKonde, 2020-02-07:
Can you confirm whether waiting for serial data to transmit via either an appropriate delay, or Serial.flush() fixes it? (note - I would not be surprised if you have to wait for one more character - the last one in the message, after Serial.flush() ) I suspect that this is the issue. 

Also, can you check if the same code works on an 328p-based board? I would expect the same behavior on the '841 as with '328p - if that's not what you're seeing, that's a bug in ATTinyCore, if you see the same problematic behavior on both, it is a bug in your code that was previously masked by a bug in the Serial code for the '841.

Sergior01, 2020-02-12:
Sorry for the late reply, been quite busy with other projects. We were prototyping the board using the Arduino IDE but we finally moved to Atmel Studio since we were also having issues configuring the Fast PWM mode on physical pin 6 (OCR2B didn't want to work with 16bit values, only 8bit, but this might be a configuration issue from our part). Everything seems fine on Atmel Studio.

I'm still using Arduino IDE for the tests and I'm going to spend some time working on the board in the next few days so I'll try your suggestions and let you know.

SpenceKonde, 2020-02-12:
That timer problem is almost certainly a bug in my core, opening new issue for that (#387) - I will have to look into this (which I would pick over the other two outstanding issues, both of which require external hardware to reproduce the reported issue). 

Man, these past two weeks everything has been blowing up on ATTinyCore, after all was quiet for a while (I mean, I had time to add a huge feature to megaTinyCore, including botching two consecutive releases wrt said new feature, without anything scary getting reported over here),

SpenceKonde, 2020-02-14:
Will investigate this weekend and see if I can reproduce

SpenceKonde, 2020-02-18:
See my reply in #387 - could it be that you were doing the same thing that I was, ie, setting OCR2B while the WGM for the timer was still set to 8-bit mode? Remember that the core sets it to 8-bit fast pwm mode before running setup, while atmel studio won't have the WGM set yet. I suspect that if, using my core, you set the WGM to a 16-bit mode before OCR2B, that would solve it. 

I have confirmed the change in behavior of Serial between 1.3.2 and 1.3.3 - on 1.3.2, serial prints are blocking - the Serial.print() will not return until all characters have actually been sent. That was a bug - correct behavior is the same as on Serial on an Uno/etc, where it  will be sent in the background unless the buffer is full.  The solution is to call Serial.flush() to wait until it's done writing all data if you need to do that (such as in your case where you need to toggle the direction pin after all sending is complete.

closed 2020-02-18 by SpenceKonde