SpenceKonde/ATTinyCore #382: Pinout Picture generator/update?

a7hybnj2, 2020-01-13:
Missing some pinout pictures for SMD versions. I would be more than happy to provide the pictures but I was hoping the pretty scheme you currently use is some kind of generated image.

Can you point me toward how to make the pinout pictures so I can add some SMD updates to your repo.

SpenceKonde, 2020-01-15:
The minigames are made by hans/@mcudude - I do not know if he has tooling
to generate them or has existing files that he modifies.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

MCUdude, 2020-01-15:
I don't have a pinout pic generator, it's actually as stupid as a Keynote project where everything is treated as figures. Not very elegant and pretty time-consuming. But the result itself looks quite nice though.

What specific SMD footprints are you missing? I'll see if I can find some time to make some more

a7hybnj2, 2020-01-15:
The one I am currently using is [ATTiny84](http://ww1.microchip.com/downloads/en/DeviceDoc/8183S.pdf) in the `QFN/MLF/VQFN` package. Luckily its probabley one of the easiest to makeup since there are only 20pins and 7 are DNC. Thanks for taking a look.

MCUdude, 2020-02-09:
@SpenceKonde can you post a list of the chips you'd like pinouts for, and what package?

MCUdude, 2020-02-09:
Is this OK?
[image]

a7hybnj2, 2020-02-09:
That certainly covers what I wanted. I already printed it out for my binder. Thanks.

MCUdude, 2020-02-09:
You can print it in landscape mode? as you can see, the "old" one above is just as long.

a7hybnj2, 2020-02-09:
[image]

As an example it doesn't look as good as the pinout for the ATTiny828 just because it is smaller on page.

Yes, I did print in landscape 😜

MCUdude, 2020-02-09:
I get your point. This would have to be something @SpenceKonde will have to decide. The good thing about the pinout I just provided is that we can get both the DIP/SOIC and the QFN one in the same pinout picture. Another alternative is to make it similar to the ATtiny828 one. A third option is to make it like this:

[image]

a7hybnj2, 2020-02-09:
Yea, totally what ever Spence wants to go with I am fine. I think these pinouts are great and I have them printed for all the AVR chips I use, not just the attinycore ones.

I like how you did the atmega324pb's with the 3 different pinout numbering schemes with 3 different pictures. So, I would be happy if the DIP and QFNs were separate images. But, again I am already happy so whatever Spence wants to go with.

Would be cool if you made a MCUdude Pinout generator somehow.

MCUdude, 2020-02-09:
> Would be cool if you made a MCUdude Pinout generator somehow.

That would indeed be really nice, but I'm not really into that sort of stuff. Ideally, it should have been a website, but I don't know much about Javascript, HTML or web design in general. What I'll do instead is a monkeys work; doing all this manually using a program that's not really meant to be used for this...

For reference, heroes what the "ATtiny828 style" would look like:
[image]

a7hybnj2, 2020-02-10:
I updated my notebook to use the latest diagonal version. It looks pretty funny with how unbalanced it is but I like it.

SpenceKonde, 2020-02-10:
Hmm, I think the one with both next to eachother looks better and strikes me as more readable, since all the text is horizontal.... I will put the other image into the repo, though. 

Thanks for the great work as always!
I wonder if there are any others that have alternate packages we haven't thought about... 

Leaving open because I am not going to deal with the images at this moment

MCUdude, 2020-02-10:
> I wonder if there are any others that have alternate packages we haven't thought about...

I don't think I'll bother creating QFN versions for absolutely _all_ ATtinys, but I can make a few more.

I'm pretty much finished with:
* ATtiny25/45/85
* ATtiny24/44/84
* ATtiny441/841

Are there other chips that "_must_" have a QFN pinout pic as well?

MCUdude, 2020-02-12:
@SpenceKonde it would be great if you'd answer...

> Are there other chips that "must" have a QFN pinout pic as well?

SpenceKonde, 2020-03-19:
84 and related 841, and the 1634 all come in a QFN/MLF package, and I have plans working their way, slowly, to bring boards based on those to market, so those are, I think the priorities. 

The 861, 167, and 43 all have QFN versions, but they are so niche, I don't think they're worth spending time on. There is also an MLF-20 version of the 85, but nobody ever seems to talk about it because it's not much smaller than the SOIC-8.

closed 2022-02-11 by SpenceKonde