SpenceKonde/ATTinyCore #385: Travis: http requests to arduino.cc sometimes fail, cause build errors

SpenceKonde, 2020-02-08:
http requests to arduino.cc seem to be failing often enough to cause about 2-4% of travis tests to fail, requiring manually restarting a bunch of jobs to get a build to pass. 

If this continues, we will need to find a way to eliminate the dependence on arduino.cc, or at least put it in only one job at the start, rather than the start of every job.

per1234, 2020-02-08:
I wonder if it could be caused by Captcha (like in https://github.com/arduino/Arduino/issues/8494)? I know that Arduino recently added some more stringent anti-DDoS measures:
https://forum.arduino.cc/index.php?topic=661642.msg4449539#msg4449539
I've been getting the occasional Captcha on the forum, which I never got before that.

SpenceKonde, 2020-02-08:
Oh interesting, that could be it.

Any thoughts on how I should work around this? should I point all the library URLs for my copies of a library to a "workaround page" on drazzy.com that has links to the official doc pages, so any human who goes there would be able to find what they wanted, while the automated testing would happily report that the page always returned 200, with a note at the top of the page explaining why I had to do this? 

If drazzy.com goes down, I wouldn't care that the travis builds were failing, because fixing that instantly becomes the number 1 priority, since it would also hose board manager installations.

SpenceKonde, 2020-02-11:
I have not been seeing this issue recur with recent commits, either... It was happening several times per test run for a while, but the past few have been fine.

closed 2020-02-11 by SpenceKonde