SpenceKonde/ATTinyCore #389: Unable to compile for Attiny841

TLINDEN, 2020-02-17:
Hi,

I am using the ATTinyCore together with Arduino-Makefile. I can compile for attiny2313 or attiny85 but not for 841:
```
 #error "This version of the Arduino IDE is not supported for this part, upgrade Arduino 1.8.6 or higher or downgrade ATTinyCore to 1.2.1"
```

Arduino version is 1.8.10, ATTinyCore is latest. If I downgrade to 1.2.1, boards.txt cannot be found:
```
/usr/local/arduino/Arduino-Makefile/Arduino.mk:440: *** Currently BOARDS_TXT='./hardware/ATTinyCore121/avr/boards.txt', which is not an existing file or an invalid filename..  Stop.
```

Configuration:
```
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /usr/local/arduino 
- [USER]               ARDMK_DIR = /usr/local/arduino/Arduino-Makefile 
- [AUTODETECTED]       ARDUINO_VERSION = 1810 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [USER]               ARDUINO_SKETCHBOOK = . 
- [USER]               ALTERNATE_CORE = ATTinyCore 
- [COMPUTED]           ALTERNATE_CORE_PATH = ./hardware/ATTinyCore/avr  (from ARDUINO_SKETCHBOOK and ALTERNATE_CORE)
- [COMPUTED]           ARDUINO_VAR_PATH = ./hardware/ATTinyCore/avr/variants (from ALTERNATE_CORE_PATH)
- [COMPUTED]           BOARDS_TXT = ./hardware/ATTinyCore/avr/boards.txt (from ALTERNATE_CORE_PATH)
- [USER]               AVR_TOOLS_DIR = /usr/local/avr 
- [COMPUTED]           ARDUINO_LIB_PATH = /usr/local/arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /usr/local/arduino/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = ./libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_SUB = 841 
- [USER]               BOARD_TAG = attinyx41 
- [COMPUTED]           CORE = tinymodern (from build.core)
- [COMPUTED]           VARIANT =  (from build.variant)
- [COMPUTED]           OBJDIR = build-attinyx41-841 (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = ./hardware/ATTinyCore/avr/cores/tinymodern (from ALTERNATE_CORE_PATH, BOARD_TAG and boards.txt)
-  [USER]              MONITOR_BAUDRATE = 115200 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [USER]               CFLAGS_STD = -std=gnu11 
- [USER]               CXXFLAGS_STD = -std=gnu++11 
- [COMPUTED]           DEVICE_PATH = /dev/ttyACM0 (from MONITOR_PORT)
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: Basic (not AVR-aware)
- [COMPUTED]           BOOTLOADER_PARENT = /usr/local/arduino/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.6 
- [COMPUTED]           CC_VERSION = 8.3.0 (avr-gcc)
```

What am I missing?

TLINDEN, 2020-02-17:
PS: Makefile can be viewed here: https://github.com/TLINDEN/attinycore-makefile-tests/tree/master/841-tinycore-test

Everything is manually compiled, latest versions and installed to:
```
ATTinyCore   /usr/local/arduino/hardware/ATTinyCore  (symlinked to project dir)
Arduino      /usr/local/arduino
avrdude      /usr/local/bin/avrdude
avr-gcc      /usr/local/avr/bin/avr-gcc
```

TLINDEN, 2020-02-17:
Ok, this: https://github.com/TLINDEN/attinycore-makefile-tests/commit/5e2e3603e96c27515ce4206c4051cd974f08c776 fixes this. The variable ARDUINO_VERSION generated by the Arduino-Makefile doesn't conform to the format expected by ATTinyCore:

Expected: 10810
Computed: 1810

When I manually set the versoin to 10810, everything works fine. I'll open an issue there.

SpenceKonde, 2020-02-17:
Much thanks for the additional investigation effort here, glad you have sorted out what the issue was. Is this a bug in the makefile you were using? If so, has it been reported to the maintainer of the makefile? In that case, can you link to the issue here so anyone else who encounters it can find the information they need.

Let me know if there's anything I need to do on this (reopen if so). Is there a way to detect when the makefile is being used to build instead of the IDE, so that I can skip that test? I don't think the issue that the check is there to get around is relevant to makefile users in general (the problem was that the IDE was concatenating all the submenu options and putting that in the filename of the compiled core to cache it - but with enough menu options, this surpassed the filename length limits on windows. In 1.8.6, they fixed this by using a hash of the menu options, rather than a human-readable list that would get too long.

closed 2020-02-17 by SpenceKonde

TLINDEN, 2020-02-18:
> Let me know if there's anything I need to do on this (reopen if so). Is there a way to detect when the makefile is being used to build instead of the IDE, so that I can skip that test?

Yes, there's a variable `ARDMK_VERSION` but I don't know if it's available through all stages.