SpenceKonde/ATTinyCore #392: Cannot compile Wire/I2C for Attiny841

TLINDEN, 2020-02-27:
I want to use an external library which uses the Wire library on an Attiny841. The 841 has I2C support via USI, but it doesn't compile:

```
scip@nubic: % make
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /usr/local/arduino 
- [USER]               ARDMK_DIR = /usr/local/arduino/Arduino-Makefile 
- [USER]               ARDUINO_VERSION = 10810 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [USER]               ARDUINO_SKETCHBOOK = . 
- [USER]               ALTERNATE_CORE = ATTinyCore 
- [COMPUTED]           ALTERNATE_CORE_PATH = ./hardware/ATTinyCore/avr  (from ARDUINO_SKETCHBOOK and ALTERNATE_CORE)
- [COMPUTED]           ARDUINO_VAR_PATH = ./hardware/ATTinyCore/avr/variants (from ALTERNATE_CORE_PATH)
- [COMPUTED]           BOARDS_TXT = ./hardware/ATTinyCore/avr/boards.txt (from ALTERNATE_CORE_PATH)
- [USER]               AVR_TOOLS_DIR = /usr/local/avr 
- [COMPUTED]           ARDUINO_LIB_PATH = /usr/local/arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /usr/local/arduino/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = ./libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_SUB = 841 
- [USER]               BOARD_TAG = attinyx41 
- [COMPUTED]           CORE = tinymodern (from build.core)
- [COMPUTED]           VARIANT =  (from build.variant)
- [COMPUTED]           OBJDIR = build-attinyx41-841 (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = ./hardware/ATTinyCore/avr/cores/tinymodern (from ALTERNATE_CORE_PATH, BOARD_TAG and boards.txt)
-  [USER]              MONITOR_BAUDRATE = 115200 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [USER]               CFLAGS_STD = -std=gnu11 
- [USER]               CXXFLAGS_STD = -std=gnu++11 
- [COMPUTED]           DEVICE_PATH = /dev/ttyACM0 (from MONITOR_PORT)
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: Basic (not AVR-aware)
-
-                      ARDUINO_LIBS =
- [USER]                 TinyBME280
- [USER]                 Wire
- [COMPUTED]           BOOTLOADER_PARENT = /usr/local/arduino/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.6 
- [COMPUTED]           CC_VERSION = 8.3.0 (avr-gcc)
-------------------------
mkdir -p build-attinyx41-841
/usr/local/avr/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -D__PROG_TYPES_COMPAT__ -mmcu=attiny841 -DF_CPU=8000000L -DARDUINO=10810 -DARDUINO_ARCH_AVR -I./hardware/ATTinyCore/avr/cores/tinymodern -I./hardware/ATTinyCore/avr/cores/tinymodern/api -I./hardware/ATTinyCore/avr/variants/     -I./libraries/TinyBME280   -I./libraries/Wire/src -Wall -ffunction-sections -fdata-sections -Os -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto -fno-devirtualize -fdiagnostics-color=always SensorSerial.ino -o build-attinyx41-841/SensorSerial.ino.o
/usr/local/avr/bin/avr-gcc -MMD -c -D__PROG_TYPES_COMPAT__ -mmcu=attiny841 -DF_CPU=8000000L -DARDUINO=10810 -DARDUINO_ARCH_AVR -I./hardware/ATTinyCore/avr/cores/tinymodern -I./hardware/ATTinyCore/avr/cores/tinymodern/api -I./hardware/ATTinyCore/avr/variants/     -I./libraries/TinyBME280   -I./libraries/Wire/src -Wall -ffunction-sections -fdata-sections -Os -std=gnu11 -flto -fno-fat-lto-objects -fdiagnostics-color=always hardware/ATTinyCore/avr/cores/tinymodern/pins_arduino.c -o build-attinyx41-841/core/pins_arduino.c.o
/usr/local/avr/bin/avr-gcc -MMD -c -D__PROG_TYPES_COMPAT__ -mmcu=attiny841 -DF_CPU=8000000L -DARDUINO=10810 -DARDUINO_ARCH_AVR -I./hardware/ATTinyCore/avr/cores/tinymodern -I./hardware/ATTinyCore/avr/cores/tinymodern/api -I./hardware/ATTinyCore/avr/variants/     -I./libraries/TinyBME280   -I./libraries/Wire/src -Wall -ffunction-sections -fdata-sections -Os -std=gnu11 -flto -fno-fat-lto-objects -fdiagnostics-color=always hardware/ATTinyCore/avr/cores/tinymodern/WInterrupts.c -o build-attinyx41-841/core/WInterrupts.c.o
/usr/local/avr/bin/avr-gcc -MMD -c -D__PROG_TYPES_COMPAT__ -mmcu=attiny841 -DF_CPU=8000000L -DARDUINO=10810 -DARDUINO_ARCH_AVR -I./hardware/ATTinyCore/avr/cores/tinymodern -I./hardware/ATTinyCore/avr/cores/tinymodern/api -I./hardware/ATTinyCore/avr/variants/     -I./libraries/TinyBME280   -I./libraries/Wire/src -Wall -ffunction-sections -fdata-sections -Os -std=gnu11 -flto -fno-fat-lto-objects -fdiagnostics-color=always hardware/ATTinyCore/avr/cores/tinymodern/wiring_analog.c -o build-attinyx41-841/core/wiring_analog.c.o
/usr/local/avr/bin/avr-gcc -MMD -c -D__PROG_TYPES_COMPAT__ -mmcu=attiny841 -DF_CPU=8000000L -DARDUINO=10810 -DARDUINO_ARCH_AVR -I./hardware/ATTinyCore/avr/cores/tinymodern -I./hardware/ATTinyCore/avr/cores/tinymodern/api -I./hardware/ATTinyCore/avr/variants/     -I./libraries/TinyBME280   -I./libraries/Wire/src -Wall -ffunction-sections -fdata-sections -Os -std=gnu11 -flto -fno-fat-lto-objects -fdiagnostics-color=always hardware/ATTinyCore/avr/cores/tinymodern/wiring.c -o build-attinyx41-841/core/wiring.c.o
/usr/local/avr/bin/avr-gcc -MMD -c -D__PROG_TYPES_COMPAT__ -mmcu=attiny841 -DF_CPU=8000000L -DARDUINO=10810 -DARDUINO_ARCH_AVR -I./hardware/ATTinyCore/avr/cores/tinymodern -I./hardware/ATTinyCore/avr/cores/tinymodern/api -I./hardware/ATTinyCore/avr/variants/     -I./libraries/TinyBME280   -I./libraries/Wire/src -Wall -ffunction-sections -fdata-sections -Os -std=gnu11 -flto -fno-fat-lto-objects -fdiagnostics-color=always hardware/ATTinyCore/avr/cores/tinymodern/wiring_digital.c -o build-attinyx41-841/core/wiring_digital.c.o
/usr/local/avr/bin/avr-gcc -MMD -c -D__PROG_TYPES_COMPAT__ -mmcu=attiny841 -DF_CPU=8000000L -DARDUINO=10810 -DARDUINO_ARCH_AVR -I./hardware/ATTinyCore/avr/cores/tinymodern -I./hardware/ATTinyCore/avr/cores/tinymodern/api -I./hardware/ATTinyCore/avr/variants/     -I./libraries/TinyBME280   -I./libraries/Wire/src -Wall -ffunction-sections -fdata-sections -Os -std=gnu11 -flto -fno-fat-lto-objects -fdiagnostics-color=always hardware/ATTinyCore/avr/cores/tinymodern/wiring_pulse.c -o build-attinyx41-841/core/wiring_pulse.c.o
/usr/local/avr/bin/avr-gcc -MMD -c -D__PROG_TYPES_COMPAT__ -mmcu=attiny841 -DF_CPU=8000000L -DARDUINO=10810 -DARDUINO_ARCH_AVR -I./hardware/ATTinyCore/avr/cores/tinymodern -I./hardware/ATTinyCore/avr/cores/tinymodern/api -I./hardware/ATTinyCore/avr/variants/     -I./libraries/TinyBME280   -I./libraries/Wire/src -Wall -ffunction-sections -fdata-sections -Os -std=gnu11 -flto -fno-fat-lto-objects -fdiagnostics-color=always hardware/ATTinyCore/avr/cores/tinymodern/wiring_shift.c -o build-attinyx41-841/core/wiring_shift.c.o
/usr/local/avr/bin/avr-g++ -MMD -c -D__PROG_TYPES_COMPAT__ -mmcu=attiny841 -DF_CPU=8000000L -DARDUINO=10810 -DARDUINO_ARCH_AVR -I./hardware/ATTinyCore/avr/cores/tinymodern -I./hardware/ATTinyCore/avr/cores/tinymodern/api -I./hardware/ATTinyCore/avr/variants/     -I./libraries/TinyBME280   -I./libraries/Wire/src -Wall -ffunction-sections -fdata-sections -Os -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto -fno-devirtualize -fdiagnostics-color=always hardware/ATTinyCore/avr/cores/tinymodern/abi.cpp -o build-attinyx41-841/core/abi.cpp.o
/usr/local/avr/bin/avr-g++ -MMD -c -D__PROG_TYPES_COMPAT__ -mmcu=attiny841 -DF_CPU=8000000L -DARDUINO=10810 -DARDUINO_ARCH_AVR -I./hardware/ATTinyCore/avr/cores/tinymodern -I./hardware/ATTinyCore/avr/cores/tinymodern/api -I./hardware/ATTinyCore/avr/variants/     -I./libraries/TinyBME280   -I./libraries/Wire/src -Wall -ffunction-sections -fdata-sections -Os -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto -fno-devirtualize -fdiagnostics-color=always hardware/ATTinyCore/avr/cores/tinymodern/HardwareSerial0.cpp -o build-attinyx41-841/core/HardwareSerial0.cpp.o
/usr/local/avr/bin/avr-g++ -MMD -c -D__PROG_TYPES_COMPAT__ -mmcu=attiny841 -DF_CPU=8000000L -DARDUINO=10810 -DARDUINO_ARCH_AVR -I./hardware/ATTinyCore/avr/cores/tinymodern -I./hardware/ATTinyCore/avr/cores/tinymodern/api -I./hardware/ATTinyCore/avr/variants/     -I./libraries/TinyBME280   -I./libraries/Wire/src -Wall -ffunction-sections -fdata-sections -Os -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto -fno-devirtualize -fdiagnostics-color=always hardware/ATTinyCore/avr/cores/tinymodern/HardwareSerial1.cpp -o build-attinyx41-841/core/HardwareSerial1.cpp.o
[... cut ...] inyx41-841/libcore.a -lc -lm 
/usr/local/avr/lib/gcc/avr/8.3.0/../../../../avr/bin/ld: /tmp/ccqfn6J0.ltrans0.ltrans.o: in function `main':
<artificial>:(.text.startup+0x118): undefined reference to `TwoWire::begin()'
collect2: error: ld returned 1 exit status
make: *** [build-attinyx41-841/841-tinycore-micronucleus-bme280-test.elf] Error 1
```

Source can be found here: https://github.com/TLINDEN/attinycore-makefile-tests/tree/master/841-tinycore-micronucleus-bme280-test. 

Do you have any idea?

SpenceKonde, 2020-02-27:
The 841 does not have a usi. It has slave-only hardware i2c, and if you
need an i2c master, you need a full software implementation. I have been
sick and have not been able to look into the many issues reported recently
with my core this week

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2020-02-27:
Though, if using the Wire library provided by the core and appropriate
tools -> wire mode submenu option, a working implementation of i2c is
supposed to be used for you!

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2020-02-27:
I do notice that you dont seem to be passing the wire mode define, though,
which would explain why Wire.h is not working...

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

TLINDEN, 2020-02-27:
Ok, which define is needed exactly? I can't find any reference to it in the readme.

And get well soon!

best,
Tom

TLINDEN, 2020-02-29:
Ok, I got it to work by using SPI. However, now I've got another problem: If I add any more code to my sample besides the measurement (e.g. some led blinking), I get:
```
/usr/local/avr/bin/avr-size build-attinyx41-841/841-tinycore-micronucleus-bme280-test.hex
   text	   data	    bss	    dec	    hex	filename
      0	   6570	      0	   6570	   19aa	build-attinyx41-841/841-tinycore-micronucleus-bme280-test.hex
...
########### Press RESET on the Nanite! ##############
sudo micronucleus build-attinyx41-841/841-tinycore-micronucleus-bme280-test.hex
> Please plug in the device ... 
> Press CTRL+C to terminate the program.
> Device is found!
connecting: 40% complete
> Device has firmware version 2.3
> Device signature: 0x1e9315 
> Available space for user applications: 6522 bytes
> Suggested sleep time between sending pages: 7ms
> Whole page count: 408  page size: 16
> Erase function sleep duration: 714ms
parsing: 60% complete
> Program file is 48 bytes too big for the bootloader!
make: *** [install] Error 1
```

Do you have any idea how to reduce the firmware size?

SpenceKonde, 2020-03-01:
You need to define one of these:
WIRE_SLAVE_ONLY
WIRE_BOTH
WIRE_MASTER_ONLY

Uhm - waitaminute - "micronucleus"? That's not my core, you're using some other core that uses that half-assed vusb bootloader.... Why people have this obsession with throwing flash away to use a flaky bitbanged-usb bootloader is beyond me. I recommend using either ISP programming, or - if you need serial uploads, say, for development convenience, a serial adapter + the serial bootloader included with my core (which only uses 640b instead of the 1600-ish that that bootloader appears to)

The other cores do lots of things differently from how my core does it (for example, they don't have the universal wire library that mine does), and nothing I say should be generalized to other cores.  If you're using someone elses core, that would explain why you were having so much trouble with Wire!

TLINDEN, 2020-03-01:
Well, no :) I'm using a watterott nanite which has the micronucleus bootloader installed, for easy development. But I'm using your core for my firmware, so that I can switch to ISP-based upload anytime. In fact I can just switch the mcu in the Makefile (see subdirectory 85-tinycore-bme280-test), recompile and upload to an attiny85 using an ISP.

> I recommend using either ISP programming, or - if you need serial uploads, say, for development convenience, a serial adapter + the serial bootloader included with my core

You're talking about Optiboot arent't you? I'll take a look into it and try it out.

SpenceKonde, 2020-03-01:
Aaaah, I see. 

Without seeing the code, I couldn't make any suggestions on size reduction. There are a bunch of "usual suspects" to look for, but I don't have a list of them anywhere. Off the top of my head, using floating point numbers is bad (all the math is implemented in software), if you're using anything that uses malloc()/free() (dynamic memory allocation), those functions eat a bunch of space too (and open up cans of worms w/regards to stability and predictability). You can always save a bit of space by excising all instances of digitalWrite/pinMode/etc and replacing with direct port writes, too... (that bit about floating point numbers is one of the things that incenses me about many libraries - okay, how cool, you give me (insert measurement here) as a floating point number, cute - but in doing so, the variables take up more memory, and you have to pull in all the floating point handling, when you could just as well have used integers (eg, instead of 123.45, 12345) and adjusted for that when outputting it (eg, print 12345/100, a ., then 12345%100), using a bunch less flash and executing faster...)

Yeah, am referring to optiboot. When I'm developing with the '841, I usually use them with optiboot (and the DTR reset trick), since that gives you the convenience of serial monitor for debugging and serial uploads on the same port, just like a typical Arduino board.

The "sledgehammer" for finding ways to reduce size, of course, is to dump the assembly listing (if using the Arduino IDE, this is dumped automatically in the sketch directory when you do "export compiled binary" - for my cores, that is, the official ones don't do this).. You can check the platform.txt and batch files included with the core to get the exact syntax to use for this. You don't have to be able to understand assembly to see which functions have the biggest blocks of code associated with them, and go from there. That's how I do most of the size optimization for the core.

TLINDEN, 2020-03-02:
Thank you for the response. I managed to reduce the size enough so I could even add a RF library.  I still need to install optiboot, especially using the same port upload and serial. Thanks a lot for all your help. Oh and btw - I am the guy in northern europe who purchased a bunch of proto boards on tindie from you (for this project!). I love these pcb's, so since we're in contact here: thanks for making them :)

SpenceKonde, 2020-03-02:
Oh, great! Glad to hear it! Thanks for your order! :-)

SpenceKonde, 2020-03-02:
Did you see the attiny841 breakout boards I sell, by the way? 
Bare board https://www.tindie.com/products/drazzy/attiny84184-breakout-wserial-header-bare-board/ 
Assembled https://www.tindie.com/products/drazzy/attiny841-dev-board-woptiboot-assembled/

Has the DTR reset circuit and such...

SpenceKonde, 2020-03-02:
Where is the makefile you're using from? I know there was a makefile that supported ATTinyCore, but I don't think it's been updated since a lot of new menu options were added.

TLINDEN, 2020-03-02:
Yes, I've seen them, but I have already ordered a couple of watterot Nanites. The Makefile I'm using is from https://github.com/sudar/Arduino-Makefile.

SpenceKonde, 2020-04-08:
Yeah - that's the one. It sounds like it is in dire need of updating to accommodate the new menu options.

closed 2020-04-08 by SpenceKonde