SpenceKonde/ATTinyCore #394: Serve package_drazzy.com_index.json with https

Legion2, 2020-03-21:
The currently used url is http only: http://drazzy.com/package_drazzy.com_index.json

It's 2020 please enable https.

SpenceKonde, 2020-04-06:
This is not a priority - I have a couple of huge development projects in progress that I am racing to progress on to get a first-mover advantage. If it is as simple as a series of commands I type in (I hear lets encrypt may be like that?), if you send me the series of commands to copy/paste into my server (it's Amazon Linux on EC2 running vanilla, fully updated, Apache 2.4), I'll copy/paste commands - but barring that kind of assistance, this will not be changed until some other change causes a critical failure that forces me to drop everything to address it.

SpenceKonde, 2020-04-06:
And to clarify - my skills with systems administration are nowhere near my skills at embedded device programming. I don't have a clue how to set this up with a real cert - I've only set up servers with self-signed certs for testing at my day job.

closed 2020-04-06 by SpenceKonde

Legion2, 2020-04-06:
Setting up a cert with letsencrypt is very simple, but you should not simply copy/paste commands without knowing the consequences and what you are doing.

PS: if you only host a public available static website on EC2, it's much cheaper (free) to host the site on GitHub pages. (which has https enabled by default)