SpenceKonde/ATTinyCore #396: ATTiny841 Timing

rayroutledge, 2020-03-25:
I programmed an ATTiny841 chip i had laying around a couple of months ago, bought some more chips and programmed them and my code did not look to work. Going back and trying my code on the original chip that one is no longer working either.

After some testing i believe this is down to timings, when using delayMicroseconds(10000) this actually takes around 77788 on average to complete. If i change the clock to 4Mhz or 1Mhz internal then they are around the 10000 expected without calibrating the OSCCAL for my voltage of 5v.
Using 8Mhz i am unable to use hardware serial.

Since I programmed the first chip I have updated both Arduino to 1.8.12 and ATTinyCore to 1.3.4

SpenceKonde, 2020-04-08:
Cannot reproduce issue with delayMicroseconds. delayMicroseconds(10000) for me consistently produces a delay of 10ms as expected. It is highly likely that you forgot to do "burn bootloader" which is necessary to set the clock speed, and were using the parts at the default-from-the-factory fuse settings of 1MHz. You must "burn bootloader" after changing board options that impact fuse settings (EESAVE, clock speed/source, BOD) in order to set those fuses.

The behavior of the internal oscillator at 5V operating frequency is called out multiple times in the core documentation. The 841's (and 1634, which uses same oscillator) is consistently around 5% fast at 4v (as opposed to the rest of the classic AVRs, where the effect of voltage on oscillator speed is much lower, but part-to-part variation and dependence on temperature is stronger). Although the core includes a special bootloader for 8MHz internal at >4v, as stated in documentation, this only provides for the bootloader (it's just compiled for 8.4MHz). To get working serial at 5v on the internal oscillator, you either need to change OSCCAL or specify a baud rate around 5% slower than the target baud rate. We recommend using a crystal when using these parts at 5v whenever timing is critical, including when making use of the serial ports.

closed 2020-04-08 by SpenceKonde

rayroutledge, 2020-04-08:
I ended up doing this from scratch with registers in MPLab, but what you say makes sense, was not something i remember seeing that needed to be done, just must have done it on that first chip and not realised.

Thank you for the feedback, will remember that for future tiny core projects.

rayroutledge, 2020-04-08:
Well i will be dammed
"If this is the first time you are using a brand new chip, **or if you need to change the clock source or BOD settings, you must do "Burn Bootloader" to set the fuses**. If you previously used optiboot, and wish to program with ISP instead, or if you want to install optiboot on the chip, you must do "burn bootloader" first as well."