SpenceKonde/ATTinyCore #397: Remapping pins used for Serial.Print

JohnOH, 2020-04-06:
I have some mature hardware  and need to map the Software Serial port as follows:
Connect Tiny84 PA3 (D7) to USB-BUB RXD for serial output from sketch.
Connect Tiny84 PA2 (D8) to USB-BUB TXD for serial input to sketch.

Do you have any pointer to where I might find the appropriate files?

SpenceKonde, 2020-04-06:
Use SoftwareSerial library. It should work and is included with the core. While the TX pin on the builtin software serial "Serial" can be moved (though the core does not expose this because I can't think of a good way to implement it - actually, I just got some ideas - but currently megaTinyCore 2.0.0 is the priority, after which it's my secret project, because I want to get first mover advantage on Arduino IDE support for a very promising family of parts), the RX is fixed by hardware because it uses the ACO interrupt, which is only present on one pin, to provide RX functionality without taking every single PCINT.

So if you need software serial on other pins, you need to use SoftwareSerial, instead of the builtin soft serial Serial.

I will open an issue for including "port-specific" copies of SoftwareSerial which will only work on a given "port" - but which in turn don't boggart every single PCINT....

SpenceKonde, 2020-04-06:
Closing as this does not represent a defect in the core but a hardware limitation. Feel free to ask further questions here - I will still get notifications.

closed 2020-04-06 by SpenceKonde

JohnOH, 2020-04-07:
Please help me understand this a little further. I currently think that the RX pin is fixed by hardware to AIN1 (PA2).
Whereas the TX pin is in software currently using AIN0 (PA1). Please will you advise which file has the AIN0 definition as a TX output.

SpenceKonde, 2020-04-07:
https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/variants/tinyX4/pins_arduino.h

```
#define ANALOG_COMP_DDR               DDRA
#define ANALOG_COMP_PORT              PORTA
#define ANALOG_COMP_PIN               PINA
#define ANALOG_COMP_AIN0_BIT          1
#define ANALOG_COMP_AIN1_BIT          2
```

The builtin software serial is the only thing that uses those defines - there's no builtin analog comparator library.

JohnOH, 2020-04-07:
Changing #define ANALOG_COMP_AIN0_BIT          1 
to value 3 moves the transmitted data pin as you said.

Many thanks.