SpenceKonde/ATTinyCore #401: Serial baudrate getting off

Sasszem, 2020-05-06:
### My setup
- 4.9V power supply trough USBASP clone
- breadboard construction
- .1uF film capacitor between VCC and GND
- FTDI breakout as serial
- Arduino 1.8.12 on Ubuntu
- version 1.3.3 of this library installed
- Attny24 with 8MHz internal oscillator
- set baud rate to 9600, 4800
### Problem
- measured baud rate 2495 (with Saleale clone logic analyzer)
- highly sketch-dependent
- Arduino IDE can still receive if serial monitor baudrate is set to 2400
- no problem if setting the baud rate in sketch to 2400 or lower 

### My sketch
```
void setup() {
  Serial.begin(2400);
  Serial.println("Started");
}

int maxVal = 5;
unsigned long t = 0;
void loop() {
  if (t<=millis()) {
    Serial.println(maxVal, 10);
    maxVal = 0; // LINE 1
    t = millis() + 500; // LINE 2
  }
  maxVal = 20; // LINE 3
}
```

SW-dependent:
 - removing marked line 1 or 3 alone does not affect the problem
 - removing marked line 2 solves the problem
 - removing ANY 2 of the 3 marked lines solves the problem (checked all cominations)
 - removing all 3 also solves the problem

It took me a night to figure this out, and I frankly have no damn idea what could possibly cause this.

SpenceKonde, 2020-06-18:
Wacky...  And it effects the "Started" print, I assume?
I don't suppose you could do "export compiled binary" when it both did and did not manifest this, and send me the two .lst files? (I am inclined to suspect that it is ALSO compiler version dependent, and probably only manifests on the t24, which I don't happen to have any of on hand...)

SpenceKonde, 2020-06-18:
Also, what baud rate do you measure when the problem isn't occurring? (wondering if it's generally right on the edge due to the internal oscillator being a bit too fast too)

Sasszem, 2020-06-18:
I've gave up that project for other reasons, but I'll gather all the stuff again for a test.

Sasszem, 2020-06-18:
I just see I uploaded wrong code, I used `Serial.begin(9600)` when I got the problem, with `Serial.begin(2400)` it works fine.

Sasszem, 2020-06-18:
Ok, here are some findings. I measured bit time and calculated baudrate for a few cases. I also recorded the startup and examined that.
## without removing anything
#### Timing: 

401.7 us -> 2489.42 baud (works with 2400)
#### Startup
"S" and "t" from "Started" is OK, in 9600 bauds, next character looks "x" (9600 bauds) with framing error (no stop bit) or "a" (2400) - looks like it changed baudrate in between "t" and "a".
## L1 removed
#### Timing
401.7 us -> 2489.42 bauds (works with 2400)
#### Startup
"S" and "t" are fine, "a" has the same problem descripbed above
## L3 removed
#### Timing
402-402.6 us -> 2483.85-2487.56 bauds (works with 2400)
#### Startup
Same as described above
## L2 removed
#### Timing:
107.9-108.1 us per bit -> 9267.84-9250.69 baud (works with 9600)
#### Startup
"Started" prints just fine
## L1-L3 removed
#### Timing
107.7 us -> 9285.05 bauds (works with 9600)
#### Startup
Prints just fine

[Assembly listings](https://github.com/SpenceKonde/ATTinyCore/files/4801014/serialtest.ino.zip)
(name contains line numbers NOT removed)

SpenceKonde, 2020-06-18:
What are you using as the serial adapter, may I ask?

SpenceKonde, 2020-06-18:
Also, I am confused about which baud rates you see the problem with, and which you don't...

Sasszem, 2020-06-18:
> What are you using as the serial adapter, may I ask?

A cheapo FT232 breakout from china, [this model](https://www.amazon.ae/Anmbest-Breakout-FT232RL-Adapter-Converter/dp/B07CQXLK17) (I bought mine from ebay)

How would this affect the baudrate?

> Also, I am confused about which baud rates you see the problem with, and which you don't...

I have problems with the first 3 cases, and the last 2 work fine (altrought their timing is still a bit off). What works with 9600 bauds is OK for me, but the 2400 variants do not work as expected.

BTW I'm using **`Arduino 1.8.5` (`avr-gcc 7.3.0`)**

SpenceKonde, 2020-06-18:
What the fu.... okay, this makes sense. 
2400 is too slow - the tuned assembly delay loop would need to run through more than 255 iterations, but the datatype is a byte. 

There's a check for this:

```
void TinySoftwareSerial::begin(long baud)
{
  long tempDelay = (((F_CPU/baud)-39)/12);
  if ((tempDelay > 255) || (tempDelay <= 0)){
  end(); //Cannot start as it would screw up uartDelay().
  }
```

But, since this is embedded, it has no way to communicate this issue to the user... and the methods that print the characters don't check if the serial has been begun (it's not event tracked  - end() just ensures that the interrupt for receiving is disabled) - and again, since there's no way it could report errors - would it be any better? 

If you do out the math, to get 2400 baud on 8MHz, the delay loop would need to run 274 times. However, since that variable never got set, it was instead at it's initial value of 0. 
so the first iteration of the loop would roll it over. and it would run 256 times. 

that would get us a baud rate of like 2590? but as we see at 9600 baud, it's coming in almost 4% slow...

Leading to the 2490 baud.... 

Frankly, what's sort of amazing is that the 9600 baud is working despite the 3.3%  baud rate error.....

I wonder whether the oscillator is kind of slow on that chip, or whether that delay calculation ought to be tweaked to give slightly shorter delays....

I will document the minimum baud rate for builtin software serial.

SpenceKonde, 2020-06-19:
In any event, I think the reason it seemed to have that strange dependence on other pieces of code was that it was balanced on a knife edge, and anything that happened to nudge the timing even a little was enough to change the behavior

Sasszem, 2020-06-19:
I'm unsure that that is the case here.
- I set my baudrate to 9600
- the chip sends data at ~2400
So it's still far slower than expected and makes debug hard.

SpenceKonde, 2020-06-19:
Wait what?!

SpenceKonde, 2020-06-19:
Okay, I totally misunderstood you above, I was wondering why you were mixing 9600 and 2400 baud rates

SpenceKonde, 2020-06-19:
But.... I do think it's getting into a state where the delay value is 0, as that is exactly the speed of serial output we'd see if it was....

nerdralph, 2020-06-25:
> 
> Frankly, what's sort of amazing is that the 9600 baud is working despite the 3.3% baud rate error.....
>

A good UART can still read data error-free with a 5% timing error.  Sampling of the 8th data bit is 8.5 bit-times after the falling edge of the start bit.  8.5 * 1.05 = 8.925, which is still within the 8th bit.

cburstedde, 2021-02-14:
> // My sketch

    if (t<=millis()) {
      Serial.println(maxVal, 10);
      maxVal = 0; // LINE 1
      t = millis() + 500; // LINE 2
    }

> // ... end quote of sketch

This sketch has the widespread issue that the time between the two `millis()` calls gets ignored. Especially the `println` call will delay the main loop timing enough to effectively destroy it. Suggesting this common solution (without getting into what `maxVal` is for):

    unsigned mlast;
    void setup() {
      Serial.begin(9600);
      Serial.println("Started");
      mlast = millis ();
    }
    int maxVal = 5;
    void loop() {
      if (millis () - mlast >= 500) {
        Serial.println(maxVal, 10);
        maxVal = 0; // LINE 1
        mlast += 500; // LINE 2
      }
      maxVal = 20; // LINE 3
    }

This may not do anything about the baud rate but fixes the println occurrences to be exactly 2 per second on average.

SpenceKonde, 2021-02-14:
Good catch, thanks - now we're down to the claimed bug that I accepted. Personally I think there is some seriously bad juju with this library. See also: https://github.com/SpenceKonde/ATTinyCore/issues/475
(in short: TIn some simple cases  the compiler is optimizing the .available() member function such that incoming characters are never recognized - but it's not optimizing away very much, because the difference in sketch size is small indeed, If there was a bug in the assembly, I'd feel like I know what I need to to get a handle on it. But I don't feel prepared to address that one.

cburstedde, 2021-02-14:
Yes, the `available()` having side effects is a bit worrying...

SpenceKonde, 2021-02-14:
Not sure I follow the trail from my description of preoblem to concluding it has sideffects

SpenceKonde, 2022-01-26:
I've just refactored the hell out of tinySoftSerial. This should be retested.