SpenceKonde/ATTinyCore #402: Issues with PWM using AnalogWrite() on ATtiny 1634

dtlepley, 2020-05-18:
Hello,

I am using a 1634 with a 12MHz external clock and am experiencing a PWM frequency of 64Hz on one PWM output and 32Hz on another (channels 2 and 3 respectively).  Both channels are PWM driving a LED driver.  This is leading to a visible blinking of the 32Hz channel.  Attached are the scope pictures of the two drive channels.
[image]

The MCU really only needs to read an input and or two and set the PWM frequency (brightness) accordingly, so no other critical timing tasks.  

Is this consistent with the behavior of the core?  What is the best path towards increasing the PWM frequency above the visible range?

-Dave

SpenceKonde, 2020-05-18:
Did you forget to "burn bootloader" to set clock source? The pwm frequency
should be 12x that, which suggests that it is still running at 1 MHz . Per
docs and in keeping with conventions set by official cores, fuses are only
at on "burn bootloader", never on normal upload.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

dtlepley, 2020-05-18:
Well now I feel stupid...I was programming directly with a USBtinyISP and totally brain
farted...I was pulling my hair out!

All working now.

Really appreciate the contribution of the ATtiny core and the support.

Thanks!
-Dave

SpenceKonde, 2020-06-02:
Thanks! Closing.

closed 2020-06-02 by SpenceKonde