SpenceKonde/ATTinyCore #404: [Question] wake from sleep mode by I2C master

klerone, 2020-05-25:
Hello SpenceKonde

I am a hobbyist and I am planning a custom board using an attiny44 as a helper in the I2C bus, it will be sleeping and Needs to wake on master request to do its job then back to sleep.

The question is, is it possible with the wire library you supply? I read the code but I could not understand too much. 
Google help me to find a library that claims to do that, USI-Wire. But I prefer to use yours to keep my project easy. 

Thank you for your great job!

Best regards.

Francisco.

SpenceKonde, 2020-06-02:
The Wire library for USI parts pulls in a version of USIWire - so it may work. I didn't write that code, I just duct-taped three wire implementations together with #defines to switch which one is used; I have not investigated this further - currently there are outstanding issues on megaTinyCore and the upcoming DACore that are higher priority, so I cannot provide an estimate of when I will able to look at this.

liebman, 2020-06-03:
I use this on an ATTiny85 in https://github.com/liebman/AnalogClock and it wakes fine from sleep.

klerone, 2020-06-08:
> The Wire library for USI parts pulls in a version of USIWire - so it may work. I didn't write that code, I just duct-taped three wire implementations together with #defines to switch which one is [...]

Well, I will try it out on a breadboard when get some spare time.

klerone, 2020-06-09:
well, I can confirm that the wire library can wake up the attiny44 from the deep sleep state when the master in the I2C bus send a command.

In my test I programmed the attiny44 to sleep and is set to listen the onReceive() method, get a byte with the PIN number of the led, blink the red led then go to back to sleep.

I used only the Wire library that bundled with the core.

I hope my test is useful for someone else too.

[image]
[image]

closed 2020-06-17 by klerone